pybluez/pybluez #369: AttributeError: module 'bluetooth._msbt' has no attribute 'initwinsock'

neicul01, 2020-11-05:
## System

 - **Operating System:** Windows 10
 - **Hardware:**  external Bluetooth adaptor (tp-link Bluetooth 4.0 UB 400)
 - **Python Version:** 3.9.0
 - **PyBluez Version:**0.23

## Issue

The installation of pybluez sugccesses, but if I run my modul or any examples I get this Error:

> Traceback (most recent call last):
>   File "D:\UNI\Mbot\pybluez-master\examples\simple\rfcomm-client.py", line 13, in <module>
>     import bluetooth
>   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
>   File "<frozen importlib._bootstrap>", line 627, in _load_backward_compatible
>   File "<frozen zipimport>", line 259, in load_module
>   File "D:\0001 lu pc\programm\python\lib\site-packages\pybluez-0.30-py3.9-win-amd64.egg\bluetooth\__init__.py", line 33, in <module>
>     from bluetooth.msbt import *
>   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
>   File "<frozen importlib._bootstrap>", line 627, in _load_backward_compatible
>   File "<frozen zipimport>", line 259, in load_module
>   File "D:\0001 lu pc\programm\python\lib\site-packages\pybluez-0.30-py3.9-win-amd64.egg\bluetooth\msbt.py", line 4, in <module>
>     bt.initwinsock ()
> AttributeError: module 'bluetooth._msbt' has no attribute 'initwinsock'
> 

I dont now how to fix the Error. 
Can anyone give me a hint ?

hiaselhans, 2020-11-07:
that function should come with the c extension being built:
https://github.com/pybluez/pybluez/blob/d7f36702bba2abc8ea41977b2eea56a17431ac6d/msbt/_msbt.c#L999

since you are using python3.9 i suggest using native bluetooth sockets instead.

`bt_socket = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_STREAM, socket.BTPROTO_RFCOMM)`

hiaselhans, 2020-11-07:
oops, sorry. seems like it already fails on import. I think that your c extension is not being built. there must have been a failure at install already?

unfortionately there are no wheels for recent python versions :/

neicul01, 2020-11-08:
ok, if i understand is correctly, it is required to install 
Microsoft Visual C++ 14.0 (have installed this ->https://visualstudio.microsoft.com/de/visual-cpp-build-tools/)
Windows 10 SDK and the c++ buildingtool are installed automaticly by installing the Microsoft Visual C++ 14.0 from this link.
Is there anythink else to install?

neicul01, 2020-11-08:
I did the installation again. Here's the result. Can't see an error

> D:\UNI\Mbot\pybluez-master>python setup.py install
> running install
> running bdist_egg
> running egg_info
> writing PyBluez.egg-info\PKG-INFO
> writing dependency_links to PyBluez.egg-info\dependency_links.txt
> writing eager_resources to PyBluez.egg-info\eager_resources.txt
> writing requirements to PyBluez.egg-info\requires.txt
> writing top-level names to PyBluez.egg-info\top_level.txt
> reading manifest file 'PyBluez.egg-info\SOURCES.txt'
> reading manifest template 'MANIFEST.in'
> warning: no files found matching 'README'
> warning: no files found matching '*.c' under directory 'macos'
> warning: no previously-included files matching '*.pyc' found under directory '*'
> warning: no previously-included files matching '*.o' found under directory '*'
> warning: no previously-included files matching '*.opp' found under directory '*'
> writing manifest file 'PyBluez.egg-info\SOURCES.txt'
> installing library code to build\bdist.win-amd64\egg
> running install_lib
> running build_py
> running build_ext
> creating build\bdist.win-amd64\egg
> creating build\bdist.win-amd64\egg\bluetooth
> copying build\lib.win-amd64-3.9\bluetooth\ble.py -> build\bdist.win-amd64\egg\bluetooth
> copying build\lib.win-amd64-3.9\bluetooth\bluez.py -> build\bdist.win-amd64\egg\bluetooth
> copying build\lib.win-amd64-3.9\bluetooth\btcommon.py -> build\bdist.win-amd64\egg\bluetooth
> copying build\lib.win-amd64-3.9\bluetooth\macos.py -> build\bdist.win-amd64\egg\bluetooth
> copying build\lib.win-amd64-3.9\bluetooth\msbt.py -> build\bdist.win-amd64\egg\bluetooth
> copying build\lib.win-amd64-3.9\bluetooth\widcomm.py -> build\bdist.win-amd64\egg\bluetooth
> copying build\lib.win-amd64-3.9\bluetooth\_msbt.cp39-win_amd64.pyd -> build\bdist.win-amd64\egg\bluetooth
> copying build\lib.win-amd64-3.9\bluetooth\__init__.py -> build\bdist.win-amd64\egg\bluetooth
> byte-compiling build\bdist.win-amd64\egg\bluetooth\ble.py to ble.cpython-39.pyc
> byte-compiling build\bdist.win-amd64\egg\bluetooth\bluez.py to bluez.cpython-39.pyc
> byte-compiling build\bdist.win-amd64\egg\bluetooth\btcommon.py to btcommon.cpython-39.pyc
> byte-compiling build\bdist.win-amd64\egg\bluetooth\macos.py to macos.cpython-39.pyc
> byte-compiling build\bdist.win-amd64\egg\bluetooth\msbt.py to msbt.cpython-39.pyc
> byte-compiling build\bdist.win-amd64\egg\bluetooth\widcomm.py to widcomm.cpython-39.pyc
> byte-compiling build\bdist.win-amd64\egg\bluetooth\__init__.py to __init__.cpython-39.pyc
> creating stub loader for bluetooth\_msbt.cp39-win_amd64.pyd
> byte-compiling build\bdist.win-amd64\egg\bluetooth\_msbt.py to _msbt.cpython-39.pyc
> creating build\bdist.win-amd64\egg\EGG-INFO
> copying PyBluez.egg-info\PKG-INFO -> build\bdist.win-amd64\egg\EGG-INFO
> copying PyBluez.egg-info\SOURCES.txt -> build\bdist.win-amd64\egg\EGG-INFO
> copying PyBluez.egg-info\dependency_links.txt -> build\bdist.win-amd64\egg\EGG-INFO
> copying PyBluez.egg-info\eager_resources.txt -> build\bdist.win-amd64\egg\EGG-INFO
> copying PyBluez.egg-info\requires.txt -> build\bdist.win-amd64\egg\EGG-INFO
> copying PyBluez.egg-info\top_level.txt -> build\bdist.win-amd64\egg\EGG-INFO
> copying PyBluez.egg-info\zip-safe -> build\bdist.win-amd64\egg\EGG-INFO
> writing build\bdist.win-amd64\egg\EGG-INFO\native_libs.txt
> creating 'dist\PyBluez-0.30-py3.9-win-amd64.egg' and adding 'build\bdist.win-amd64\egg' to it
> removing 'build\bdist.win-amd64\egg' (and everything under it)
> Processing PyBluez-0.30-py3.9-win-amd64.egg
> Removing d:\0001 lu pc\programm\python\lib\site-packages\PyBluez-0.30-py3.9-win-amd64.egg
> Copying PyBluez-0.30-py3.9-win-amd64.egg to d:\0001 lu pc\programm\python\lib\site-packages
> PyBluez 0.30 is already the active version in easy-install.pth
> 
> Installed d:\0001 lu pc\programm\python\lib\site-packages\pybluez-0.30-py3.9-win-amd64.egg
> Processing dependencies for PyBluez==0.30
> Finished processing dependencies for PyBluez==0.30
> 
> D:\UNI\Mbot\pybluez-master>
>

neicul01, 2020-11-12:
solved it! first run the 
`pip install pybluez `
than
`python setup.py install `

avtar2084, 2020-11-24:
this issue is with python 3.9 . Install python 3.8 and you will not have this issue

TheethumNandrum, 2020-12-20:
I downgraded to Python 3.8 and the error still persists... Any ideas?

BobShin88, 2021-01-03:
Try to reinstall, just OK.

pip install --upgrade --force-reinstall pybluez

closed 2021-02-13 by stale[bot]

Vijayvj1, 2023-07-04:
Still Issue is active unable to solve in windows machine