pybluez/pybluez #373: Does pybluez support ble? Does pybluez support ble on windows

Chenilove, 2020-11-21:
I saw a ble in the official example, but I didn't run successfully. Do I need Linux?

xloem, 2020-12-14:
Looking at the source, it looks like the ble support is a partial import of libgatt.  Most of the libgatt functionality is in submodules and is left out.  There's a good list of projects at https://github.com/ukBaz/python-bluezero/wiki

xloem, 2021-01-14:
@stale here closing this issue is the right thing to do, but the `wontfix` label is not correct.
it would be better if you would accept community commands on what to do to the issue.

xloem, 2021-01-14:
@stale !!!!! sorry, i made things worse by talking to you.  reminder to self: DO NOT REPLY TO STALE WHEN THEY MARK THE ISSUE FOR CLOSURE.

closed 2021-02-21 by stale[bot]