pybluez/pybluez #374: Why I can't find BLE Devices at Raspiberry4b+?

wsysx123, 2020-11-22:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:Raspbian 
 - **Hardware: Raspberry Pi 4b+
 - **Python Version:3.7.3
 - **PyBluez Version:0.23

## Issue

I Run Code by Command.This is the picture.Why the length is always 0?
[image]

dariussullivan, 2020-11-24:
Those processes `bluetoothctl` and `python3` are stopped, not killed. Are you trying to use CTRL+Z to quit? That leaves `bluetoothctl` still running, maybe with the device still open. Type `fg` and they will resume.

Use CTRL+D to quit instead.

wsysx123, 2020-11-29:
I have understood this problem because my device is BLE, not bluetooth, and needs to use different search commands. thank you very much!

closed 2020-11-29 by wsysx123