pybluez/pybluez #379: "python setup.py install" does not work on OSX (because lightaquablue)

rambo, 2021-01-07:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** osx catalina
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.) mbp
 - **Python Version:** 3.9
 - **PyBluez Version:** git d7f36702bba2abc8ea41977b2eea56a17431ac6d

## Issue

Found when debugging #378 

```
(pybluezdev) ➜  pybluez git:(issue-378) ✗ python setup.py install
running install
running bdist_egg
running egg_info
writing PyBluez.egg-info/PKG-INFO
writing dependency_links to PyBluez.egg-info/dependency_links.txt
writing eager_resources to PyBluez.egg-info/eager_resources.txt
writing requirements to PyBluez.egg-info/requires.txt
writing top-level names to PyBluez.egg-info/top_level.txt
reading manifest file 'PyBluez.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'README'
warning: no files found matching '*.c' under directory 'macos'
warning: no previously-included files matching '*.pyc' found under directory '*'
warning: no previously-included files matching '*.o' found under directory '*'
warning: no previously-included files matching '*.opp' found under directory '*'
writing manifest file 'PyBluez.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.15-x86_64/egg
running install_lib
running build_py
creating build/bdist.macosx-10.15-x86_64/egg
creating build/bdist.macosx-10.15-x86_64/egg/bluetooth
copying build/lib/bluetooth/btcommon.py -> build/bdist.macosx-10.15-x86_64/egg/bluetooth
copying build/lib/bluetooth/ble.py -> build/bdist.macosx-10.15-x86_64/egg/bluetooth
copying build/lib/bluetooth/macos.py -> build/bdist.macosx-10.15-x86_64/egg/bluetooth
copying build/lib/bluetooth/__init__.py -> build/bdist.macosx-10.15-x86_64/egg/bluetooth
copying build/lib/bluetooth/bluez.py -> build/bdist.macosx-10.15-x86_64/egg/bluetooth
copying build/lib/bluetooth/msbt.py -> build/bdist.macosx-10.15-x86_64/egg/bluetooth
copying build/lib/bluetooth/widcomm.py -> build/bdist.macosx-10.15-x86_64/egg/bluetooth
creating build/bdist.macosx-10.15-x86_64/egg/PyBluez.egg-info
copying build/lib/PyBluez.egg-info/PKG-INFO -> build/bdist.macosx-10.15-x86_64/egg/PyBluez.egg-info
copying build/lib/PyBluez.egg-info/eager_resources.txt -> build/bdist.macosx-10.15-x86_64/egg/PyBluez.egg-info
copying build/lib/PyBluez.egg-info/not-zip-safe -> build/bdist.macosx-10.15-x86_64/egg/PyBluez.egg-info
copying build/lib/PyBluez.egg-info/SOURCES.txt -> build/bdist.macosx-10.15-x86_64/egg/PyBluez.egg-info
copying build/lib/PyBluez.egg-info/requires.txt -> build/bdist.macosx-10.15-x86_64/egg/PyBluez.egg-info
copying build/lib/PyBluez.egg-info/top_level.txt -> build/bdist.macosx-10.15-x86_64/egg/PyBluez.egg-info
copying build/lib/PyBluez.egg-info/dependency_links.txt -> build/bdist.macosx-10.15-x86_64/egg/PyBluez.egg-info
creating build/bdist.macosx-10.15-x86_64/egg/lightblue
copying build/lib/lightblue/_bluetoothsockets.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
copying build/lib/lightblue/_LightAquaBlue.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
copying build/lib/lightblue/_obexcommon.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
copying build/lib/lightblue/_IOBluetoothUI.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
copying build/lib/lightblue/__init__.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
copying build/lib/lightblue/_IOBluetooth.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
creating build/bdist.macosx-10.15-x86_64/egg/lightblue/__pycache__
copying build/lib/lightblue/__pycache__/__init__.cpython-39.pyc -> build/bdist.macosx-10.15-x86_64/egg/lightblue/__pycache__
copying build/lib/lightblue/__pycache__/_IOBluetooth.cpython-39.pyc -> build/bdist.macosx-10.15-x86_64/egg/lightblue/__pycache__
copying build/lib/lightblue/__pycache__/_LightAquaBlue.cpython-39.pyc -> build/bdist.macosx-10.15-x86_64/egg/lightblue/__pycache__
copying build/lib/lightblue/__pycache__/_lightblue.cpython-39.pyc -> build/bdist.macosx-10.15-x86_64/egg/lightblue/__pycache__
copying build/lib/lightblue/_obex.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
copying build/lib/lightblue/_lightblue.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
copying build/lib/lightblue/obex.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
copying build/lib/lightblue/_macutil.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
copying build/lib/lightblue/_lightbluecommon.py -> build/bdist.macosx-10.15-x86_64/egg/lightblue
byte-compiling build/bdist.macosx-10.15-x86_64/egg/bluetooth/btcommon.py to btcommon.cpython-39.pyc
[... 39 lines cut ...]
Using /Users/rambo/.virtualenvs/pybluezdev/lib/python3.9/site-packages
Finished processing dependencies for PyBluez==0.30
(pybluezdev) ➜  pybluez git:(issue-378) ✗ python 
Python 3.9.1 (default, Dec 10 2020, 10:36:35) 
[Clang 12.0.0 (clang-1200.0.32.27)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import bluetooth
/Users/rambo/.virtualenvs/pybluezdev/lib/python3.9/site-packages/objc/_bridgesupport.py:13: UserWarning: Module lightblue was already imported from /Users/rambo/.virtualenvs/pybluezdev/lib/python3.9/site-packages/PyBluez-0.30-py3.9.egg/lightblue/__init__.py, but /Users/rambo/devel/pybluez is being added to sys.path
  import pkg_resources
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/rambo/devel/pybluez/bluetooth/__init__.py", line 38, in <module>
    from .macos import *
  File "/Users/rambo/devel/pybluez/bluetooth/macos.py", line 1, in <module>
    import lightblue
  File "/Users/rambo/.virtualenvs/pybluezdev/lib/python3.9/site-packages/PyBluez-0.30-py3.9.egg/lightblue/__init__.py", line 160, in <module>
    from ._lightblue import *
  File "/Users/rambo/.virtualenvs/pybluezdev/lib/python3.9/site-packages/PyBluez-0.30-py3.9.egg/lightblue/_lightblue.py", line 29, in <module>
    from . import _LightAquaBlue
  File "/Users/rambo/.virtualenvs/pybluezdev/lib/python3.9/site-packages/PyBluez-0.30-py3.9.egg/lightblue/_LightAquaBlue.py", line 32, in <module>
    raise ImportError("Cannot load LightAquaBlue framework, not found at " + \
ImportError: Cannot load LightAquaBlue framework, not found at /Users/rambo/.virtualenvs/pybluezdev/lib/python3.9/site-packages/PyBluez-0.30-py3.9.egg/lightblue/LightAquaBlue.framework
>>> 
(pybluezdev) ➜  pybluez git:(issue-378) ✗ 
```

hiwanz, 2021-02-01:
Operating System: osx bigsur
Hardware: (e.g., Raspberry Pi, external Bluetooth adaptor, etc.) mbp
Python Version: 3.9

```

Building wheel for pybluez (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/local/opt/python@3.9/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/47/sw37s42j6zg5wrcgzgj0qg3w0000gn/T/pip-install-eigv1zk3/pybluez_175419493d7b474cbf23dda13d9b812d/setup.py'"'"'; __file__='"'"'/private/var/folders/47/sw37s42j6zg5wrcgzgj0qg3w0000gn/T/pip-install-eigv1zk3/pybluez_175419493d7b474cbf23dda13d9b812d/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/47/sw37s42j6zg5wrcgzgj0qg3w0000gn/T/pip-wheel-1cebk0cr
       cwd: /private/var/folders/47/sw37s42j6zg5wrcgzgj0qg3w0000gn/T/pip-install-eigv1zk3/pybluez_175419493d7b474cbf23dda13d9b812d/
  Complete output (16 lines):
  Command line invocation:
      /Applications/Xcode.app/Contents/Developer/usr/bin/xcodebuild install -project macos/LightAquaBlue/LightAquaBlue.xcodeproj -scheme LightAquaBlue DSTROOT=/private/var/folders/47/sw37s42j6zg5wrcgzgj0qg3w0000gn/T/pip-install-eigv1zk3/pybluez_175419493d7b474cbf23dda13d9b812d/macos INSTALL_PATH=/ DEPLOYMENT_LOCATION=YES
  
  Build settings from command line:
      DEPLOYMENT_LOCATION = YES
      DSTROOT = /private/var/folders/47/sw37s42j6zg5wrcgzgj0qg3w0000gn/T/pip-install-eigv1zk3/pybluez_175419493d7b474cbf23dda13d9b812d/macos
      INSTALL_PATH = /
  
  xcodebuild: error: 'macos/LightAquaBlue/LightAquaBlue.xcodeproj' does not exist.
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/private/var/folders/47/sw37s42j6zg5wrcgzgj0qg3w0000gn/T/pip-install-eigv1zk3/pybluez_175419493d7b474cbf23dda13d9b812d/setup.py", line 79, in <module>
      subprocess.check_call([
    File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['xcodebuild', 'install', '-project', 'macos/LightAquaBlue/LightAquaBlue.xcodeproj', '-scheme', 'LightAquaBlue', 'DSTROOT=/private/var/folders/47/sw37s42j6zg5wrcgzgj0qg3w0000gn/T/pip-install-eigv1zk3/pybluez_175419493d7b474cbf23dda13d9b812d/macos', 'INSTALL_PATH=/', 'DEPLOYMENT_LOCATION=YES']' returned non-zero exit status 66.
  ----------------------------------------
  ERROR: Failed building wheel for pybluez```

RomanoB, 2021-02-11:
Same as @hiwanz here.

TheFinalJoke, 2021-03-27:
Found what is wrong, The in the setup.py install LightAquaBlue builds a macos package using xcodebuild. While my testing, it is failing every-time. Hopefully will have a PR up soon

rambo, 2021-03-27:
https://github.com/pybluez/pybluez/pull/381 should "solve" the compile issue, it doesn't seem to actually work though, something has changed in OSX APIs and lightaquablue was abandoned years ago

closed 2021-06-09 by stale[bot]