pybluez/pybluez #389: why is it so hard to get the modul working?

Ce345, 2021-02-01:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:**
 - **PyBluez Version:**

## Issue

Please provide information necessary to reproduce the issue, including sample
code.

If you have long logs, please post them on https://gist.github.com and link to
the Gist here.

Ce345, 2021-02-01:
why is it so hard to get the modul working?
spent already several hours just to work through a heap or stack of errors. the software is apperently not desiged for windows.
the hours spent by users to get it right surpasses the hours of the developper by far!

ash-nolan, 2021-02-02:
@Ce345 you don't appear to be raising a specific issue here.
What isn't working? What have you tried already? What does your setup look like? What errors are you getting? Can you provide example code that you believe should work, and can you show us the logs of the actual outcome?

It is generally considered rude to make an issue complaining about software that a group of mostly volunteers have provided to you and the wider Python community (in good faith and free of charge) when you yourself have not put any effort into providing specific examples of the problems you are facing.

closed 2021-06-02 by stale[bot]