pybluez/pybluez #390: no advertisable device

timtimno1, 2021-03-05:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:*Linux*
 - **Hardware:* Raspberry Pi* 
 - **Python Version:*3.7*
 - **PyBluez Version:**

## Issue

I use example code but,I get error.
I have search solution on internet ,I get add -C in /etc/systemd/system/dbus-org.bluez.servaice,but not work

Traceback (most recent call last):
  File "/home/pi/Desktop/4.py", line 20, in <module>
    profiles=[bluetooth.SERIAL_PORT_PROFILE],
  File "/usr/local/lib/python3.7/dist-packages/bluetooth/bluez.py", line 275, in advertise_service
    raise BluetoothError (*e.args)
bluetooth.btcommon.BluetoothError: no advertisable device

Felix-Jost, 2021-03-06:
I have the same Issue .....

timtimno1, 2021-03-08:
> I have the same Issue .....

This trouble is very tricky!

StephenCarboni, 2021-03-09:
I'm not an expert, but either of these commands worked for me:

- `bluetoothctl` and enter `discoverable on` OR `bluetoothctl -- discoverable on`
- `sudo hciconfig hci0 piscan`

tlaf0504, 2021-03-29:
Had the same issue on Fedora 33 using Intel AX200.
The solution was to create a group `bluetooth` (name it as you want) and change the group of `var/run/sdp` to the previously created group. Last but not least add also your user to this group.

A more detailed explaination is given [here](https://github.com/ev3dev/ev3dev/issues/274#issuecomment-74593671) .

H-Ribeiro, 2021-06-08:
Try running as **sudo**.
It worked for me.
Be advised that you might need to install missing libraries as root.

H-Ribeiro, 2021-07-09:
1 - configure bt device: **sudo nano /etc/systemd/system/dbus-org.bluez.service**
2 - locate the line starting **ExecStart** , and replace it with the following:
2.1 - **ExecStart=/usr/lib/bluetooth/bluetoothd --compat --noplugin=sap**
        **ExecStartPost=/usr/bin/sdptool add SP**

Hate9, 2021-07-21:
I'm having the same issue. I've tried every solution listed here, and it still does not work.

closed 2021-08-28 by stale[bot]

alex9099, 2023-02-14:
This is still an issue on arch linux with python3.8

cistercian86, 2023-09-17:
on Raspian Buster, I found that the systemd changes from @H-Ribeiro were necessary but not sufficient. Also needed to run
`sudo hciconfig hci0 piscan`

...as mentioned in https://stackoverflow.com/questions/37913796/bluetooth-error-no-advertisable-device