pybluez/pybluez #394: pybluez discover_devices returns an empty list - raspberry pi zero

MegaRusher, 2021-03-18:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:**
 - **PyBluez Version:**

## Issue
The method discover_devices() returns an empty list on my raspberry pi zero.
Tried also with root user (in case of any rights problem) but it's still empty.

I can perform everything with the normal pi user and bluetoothctl + a complex expect script,
but i'd prefer a 100% Pythonic way.

Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
>>> import bluetooth
>>> bluetooth.discover_devices()
[]

uname: Linux happy 5.10.17+ armv6l
Controller: B8:27:EB:59:73:88

Well that's it, too bad that i cannot use your library.
Thanks for reading, cheers
Manu

closed 2021-06-09 by stale[bot]

ArtemGordinsky, 2022-01-02:
@MegaRusher I'm facing the exact same issue, except on Pi3. Have you ever been able to get to the bottom of this?

Too bad this issue got no attention.

schodet, 2022-01-02:
It seems that a user of NXT-Python has the same issue: https://github.com/schodet/nxt-python/issues/175

dogtopus, 2022-01-02:
#404