pybluez/pybluez #397: Not working on macos bigsur

viral-sangani, 2021-04-08:
## System

 - MacOS BigSur - Version 11.2.3
 - **Hardware:** internal macbook bluetooth
 - **Python version** - 3.8.2
 -  **pybluez version** - 0.23 Latest

## Issue

I have created a new virtual env. and tried installing pybluez via `pip install --upgrade git+https://github.com/karulis/pybluez.git#egg=pybluez` this command. Installation is completed without any error but when tried to import is shows the following error.

```
ImportError: Cannot load LightAquaBlue framework, not found at /Users/viralsangani/Desktop/Projects/
opensource/venv/lib/python3.8/site-packages/lightblue/LightAquaBlue.framework
```
I tried installing LightAquaBlue from the source but it's showing an error while installing.

sensiblepuffin, 2021-04-18:
I had some success with installing lightblue from [here](https://github.com/0-1-0/lightblue-0.4), which installs LightAquaBlue to /Library/Frameworks/, and then copying /Library/Frameworks/LightAquaBlue.framework/ into the site-packages directory (beware, it's all links).

closed 2021-06-09 by stale[bot]

bwomsm1, 2023-03-14:
Hi, I am also facing this issue, I am currently using pybluez version 0.30, with Mac big Sur.
Any advice?