pybluez/pybluez #401: [Long term] Port C code to Cython

dogtopus, 2021-05-12:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** All (?)
 - **Hardware:** All
 - **Python Version:** All
 - **PyBluez Version:** N/A

## Issue

While works perfectly (?), pure-C modules does not fail nicely and by default outputs cryptic if not completely unreadable and useless traceback when they fail. Cython modules on the other hand generates the boilerplate for exception handling and can output concious error messages and tracebacks more easily with little to no extra code written by the devs which can help a lot when debugging low-level errors.

I might take a shot on this when I got time, starting from the Windows binding since looks like it breaks a lot and most of the recent issues are about Windows bindings throwing completely unreadable exceptions. Others are welcome to do this as well.

dogtopus, 2021-06-17:
@rgov Should we disable stale bot since it only seems to make issues harder to track and encourage people to double post.

dogtopus, 2021-06-18:
Or alternatively, we can probably use pure python with ctypes/cffi to achieve what we currently do with C extensions (and, if possible, probably pyobjus for macOS thus skipping lightblue entirely).

rgov, 2021-06-18:
I'm not really involved at all anymore but if you submit a pull request to change https://github.com/pybluez/pybluez/blob/master/.github/stale.yml I can merge it.

dogtopus, 2021-09-07:
@rgov I missed the last invitation. Could you please send it again?