pybluez/pybluez #402: Windows 10 PyBluez installation error

michaliskon, 2021-05-13:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** Windows 10
 - **Hardware:** PC
 - **Python Version:** 3.8.5 
 - **PyBluez Version:** N/A

## Issue

I am trying to install PyBluez but the installation creates an error.

`Python 3.8.5 (default, Sep  3 2020, 21:29:08) [MSC v.1916 64 bit (AMD64)]
Type "copyright", "credits" or "license" for more information.

IPython 7.19.0 -- An enhanced Interactive Python.

pip install pybluez
Collecting pybluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
Building wheels for collected packages: pybluez
  Building wheel for pybluez (setup.py): started
  Building wheel for pybluez (setup.py): finished with status 'error'
  Running setup.py clean for pybluez
Failed to build pybluez
Installing collected packages: pybluez
    Running setup.py install for pybluez: started
    Running setup.py install for pybluez: finished with status 'error'
Note: you may need to restart the kernel to use updated packages.
  ERROR: Command errored out with exit status 1:
   command: 'C:\ProgramData\Anaconda3\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Micha\\AppData\\Local\\Temp\\pip-install-7hwnodyg\\pybluez\\setup.py'"'"'; __file__='"'"'C:\\Users\\Micha\\AppData\\Local\\Temp\\pip-install-7hwnodyg\\pybluez\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\Micha\AppData\Local\Temp\pip-wheel-au6lfmhc'
       cwd: C:\Users\Micha\AppData\Local\Temp\pip-install-7hwnodyg\pybluez\
  Complete output (41 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.8
  creating build\lib.win-amd64-3.8\bluetooth
  copying bluetooth\ble.py -> build\lib.win-amd64-3.8\bluetooth
  copying bluetooth\bluez.py -> build\lib.win-amd64-3.8\bluetooth
  copying bluetooth\btcommon.py -> build\lib.win-amd64-3.8\bluetooth
  copying bluetooth\macos.py -> build\lib.win-amd64-3.8\bluetooth
  copying bluetooth\msbt.py -> build\lib.win-amd64-3.8\bluetooth
  copying bluetooth\widcomm.py -> build\lib.win-amd64-3.8\bluetooth
  copying bluetooth\__init__.py -> build\lib.win-amd64-3.8\bluetooth
  C:\ProgramData\Anaconda3\lib\site-packages\setuptools\lib2to3_ex.py:36: SetuptoolsDeprecationWarning: 2to3 support is deprecated. If the project still requires Python 2 support, please migrate to a single-codebase solution or employ an independent conversion process.
    warnings.warn(
  Fixing build\lib.win-amd64-3.8\bluetooth\ble.py build\lib.win-amd64-3.8\bluetooth\bluez.py build\lib.win-amd64-3.8\bluetooth\btcommon.py build\lib.win-amd64-3.8\bluetooth\macos.py build\lib.win-amd64-3.8\bluetooth\msbt.py build\lib.win-amd64-3.8\bluetooth\widcomm.py build\lib.win-amd64-3.8\bluetooth\__init__.py
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  Fixing build\lib.win-amd64-3.8\bluetooth\ble.py build\lib.win-amd64-3.8\bluetooth\bluez.py build\lib.win-amd64-3.8\bluetooth\btcommon.py build\lib.win-amd64-3.8\bluetooth\macos.py build\lib.win-amd64-3.8\bluetooth\msbt.py build\lib.win-amd64-3.8\bluetooth\widcomm.py build\lib.win-amd64-3.8\bluetooth\__init__.py
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  running build_ext
  building 'bluetooth._msbt' extension
  creating build\temp.win-amd64-3.8
  creating build\temp.win-amd64-3.8\Release
  creating build\temp.win-amd64-3.8\Release\msbt
  C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I.\port3 -IC:\ProgramData\Anaconda3\include -IC:\ProgramData\Anaconda3\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /Tcmsbt\_msbt.c /Fobuild\temp.win-amd64-3.8\Release\msbt\_msbt.obj
  _msbt.c
  msbt\_msbt.c(145): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
  msbt\_msbt.c(222): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
  msbt\_msbt.c(402): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
[... 30 lines cut ...]
    Skipping optional fixer: set_literal
    Skipping optional fixer: ws_comma
    Fixing build\lib.win-amd64-3.8\bluetooth\ble.py build\lib.win-amd64-3.8\bluetooth\bluez.py build\lib.win-amd64-3.8\bluetooth\btcommon.py build\lib.win-amd64-3.8\bluetooth\macos.py build\lib.win-amd64-3.8\bluetooth\msbt.py build\lib.win-amd64-3.8\bluetooth\widcomm.py build\lib.win-amd64-3.8\bluetooth\__init__.py
    Skipping optional fixer: buffer
    Skipping optional fixer: idioms
    Skipping optional fixer: set_literal
    Skipping optional fixer: ws_comma
    running build_ext
    building 'bluetooth._msbt' extension
    creating build\temp.win-amd64-3.8
    creating build\temp.win-amd64-3.8\Release
    creating build\temp.win-amd64-3.8\Release\msbt
    C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I.\port3 -IC:\ProgramData\Anaconda3\include -IC:\ProgramData\Anaconda3\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /Tcmsbt\_msbt.c /Fobuild\temp.win-amd64-3.8\Release\msbt\_msbt.obj
    _msbt.c
    msbt\_msbt.c(145): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
    msbt\_msbt.c(222): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
    msbt\_msbt.c(402): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
    msbt\_msbt.c(415): warning C4244: 'initializing': conversion from 'SOCKET' to 'int', possible loss of data
    msbt\_msbt.c(635): warning C4244: 'initializing': conversion from 'SOCKET' to 'int', possible loss of data
    msbt\_msbt.c(807): warning C4311: 'type cast': pointer truncation from 'HANDLE' to 'unsigned long'
    C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:C:\ProgramData\Anaconda3\libs /LIBPATH:C:\ProgramData\Anaconda3\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\um\x64" WS2_32.lib Irprops.lib /EXPORT:PyInit__msbt build\temp.win-amd64-3.8\Release\msbt\_msbt.obj /OUT:build\lib.win-amd64-3.8\bluetooth\_msbt.cp38-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.8\Release\msbt\_msbt.cp38-win_amd64.lib
    LINK : fatal error LNK1181: cannot open input file 'Irprops.lib'
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\BuildTools\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1181
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\ProgramData\Anaconda3\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Micha\\AppData\\Local\\Temp\\pip-install-7hwnodyg\\pybluez\\setup.py'"'"'; __file__='"'"'C:\\Users\\Micha\\AppData\\Local\\Temp\\pip-install-7hwnodyg\\pybluez\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Micha\AppData\Local\Temp\pip-record-vr0khmeb\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\ProgramData\Anaconda3\Include\pybluez' Check the logs for full command output.`

TiggerZ97, 2021-05-19:
Hello, just had the same problem, but you can fix it by just installing the package from the master source code (https://github.com/pybluez/pybluez/archive/master.zip), check the isntructions in https://github.com/pybluez/pybluez/blob/master/docs/install.rst Cheers!

michaliskon, 2021-05-19:
Hi @TiggerZ97, thank you for the advice. I tried it, but kept getting the same error.

The only think that worked was pip install file:C:\Users\Micha\pybluez-master.tar.gz

But when I try import PyBluez in python I keep getting:
`	import PyBluez
	Traceback (most recent call last):

	  File "<ipython-input-9-339d65f36d24>", line 1, in <module>
		import PyBluez

	ModuleNotFoundError: No module named 'PyBluez'`

I even tried closing spyder and re-opening, but I keep getting the same.

TiggerZ97, 2021-05-19:
Hey, try importing it as "import bluetooth" :)

michaliskon, 2021-05-19:
> 
> 
> Hey, try importing it as "import bluetooth" :)

Yes, of course!

Thank you :)

TiggerZ97, 2021-05-20:
Did it work?

closed 2021-06-29 by stale[bot]