pybluez/pybluez #403: Windows 10: Find service function throws OSError without any description.

mujiwizard, 2021-06-06:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**Windows 10 Home 64-bit (10.0, Build 19041)
 - **Laptop:** Dell inspiron 5559
 - **Hardware:** builtin intel wireless bluetooth 
 - **Python Version:*3.6.5*
 - **PyBluez Version:*0.23*

## Issue
bluetooth.find_service() throws an OSError exception without a description

here is the script I am using  https://gist.github.com/mujiwizard/345aff0d4cd7d789e1216c1a2aba173a

Here is the program.

python3 .\testBluetooth.py
3 devices found

Device Name: Hesh 3
Device MAC Address: D0:8A:55:7A:88:06
Services Found:
Traceback (most recent call last):
  File ".\BluetoothAPI.py", line 41, in <module>
    scan_services()
  File ".\BluetoothAPI.py", line 31, in scan_services
    services = bluetooth.find_service(address=addr)
  File "C:\Users\Mano\AppData\Local\Programs\Python\Python36\lib\site-packages\bluetooth\msbt.py", line 243, in find_service
    dresults = bt.find_service (addr, uuidstr)
OSError

dogtopus, 2021-06-16:
THis has been a problem for quite a while. See #307.

SharkXiaoYu, 2021-07-22:
I have the same problem。Here is the program： service_matches = bluetooth.find_service(uuid=uuid, address=addr)   OSError

dogtopus, 2021-07-22:
> I have the same problem。Here is the program： service_matches = bluetooth.find_service(uuid=uuid, address=addr) OSError

It should be fixed in the latest dev branch.

closed 2021-09-07 by stale[bot]