pybluez/pybluez #404: Update to Bluez5/DBus based API

dogtopus, 2021-06-17:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System: Linux**
 - **Hardware: Any** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version: 3.x**
 - **PyBluez Version: Any**

## Issue

PyBluez still uses the old SDPd API pre-Bluez 5, which is deprecated in favor of DBus-based API [since long ago](https://stackoverflow.com/questions/41290114/bluetooth-sdp-where-is-sdpd). This causes the SDP registration to be non-functional on modern Linux systems (can be seen when running e.g. the supplied rfcomm-server.py).

vik0t0r, 2021-07-18:
I'll start working on this

vik0t0r, 2021-08-03:
okay so I'm quite bad at this. It seems that on bluez5 you cannot find services as on bluez4 and that everything related to sdp is done through the org.bluez.Profile1 interface. As I understand, you need to register a Profile when doing the server paper as well as when doing the client paper. Moreover, sockets are created each time a new connection is made. This means that the concept of looking up sdp for a port to connect, or creating a socket before advertising  is no longer viable.

In addition, now python sockets support bluetooth without sdp (ej you have to give it a fixed port) which have left BluetoothSocket a bit obsolete .

On the other hand, for running a dbus interface you need a main loop, making every program that uses the module main loop agnostic  (or maybe the main loop could be implemented on a thread adding a layer of abstraction )

vik0t0r, 2021-08-03:
I don't understand which approach to take, seems like the most useful would be to expose all the dbus api through a more pythonic way, but for that pydbus is already there, doesn't it? (althought it seems a bit discontinued).

dogtopus, 2021-08-10:
@vik0t0r There's also [dasbus](https://github.com/rhinstaller/dasbus) which looks more maintained than pydbus and looks better than `dlopen()`ing libdbus f\*ckery.

Yes `BluetoothSocket` is officially obsolete on Linux. We can probably just pass it through to `AF_BLUETOOTH`. Not sure if everything will still work though so maybe proceed with caution.

dogtopus, 2021-08-10:
Hm both dasbus and pydbus might not work because

```
No support for Unix file descriptors: It is not possible to send or receive Unix file descriptors.
```

ukBaz, 2022-01-02:
I have been looking at some the issues around Python, D-Bus, and BlueZ. I certainly don't have all the answers and am sharing here in the hope that it will help move things forward for all of us. (Even if it is just to help understand that this isn't a path to go down)

I have seen issues with the various D-Bus Python bindings when trying to use them with BlueZ. As a result of this I have been looking into using the [PyGObject](https://pygobject.readthedocs.io/en/latest/index.html) bindings that most of the newer D-Bus libraries are built on. My thought was that this might be a better option to explore as the BlueZ D-Bus API requires a small subset of the D-Bus functionality. The other libraries are trying to be generic D-Bus bindings and cover the most widely used functionality.

Below is an experiment I did to create a BlueZ Serial Port Profile (SPP) Server using PyGObject.
```python
import os

from gi.repository import Gio, GLib

# Introspection data for DBus
profile_xml = """
<node>
    <interface name="org.bluez.Profile1">
        <method name="Release"/>
        <method name="NewConnection">
            <arg type="o" name="device" direction="in"/>
            <arg type="h" name="fd" direction="in"/>
            <arg type="a{sv}" name="fd_properties" direction="in"/>
        </method>
        <method name="RequestDisconnection">
            <arg type="o" name="device" direction="in"/>
        </method>
    </interface>
</node>
"""

class DbusService:
    """Class used to publish a DBus service on to the DBus System Bus"""
    def __init__(self, introspection_xml, publish_path):
        self.node_info = Gio.DBusNodeInfo.new_for_xml(introspection_xml).interfaces[0]
        # start experiment
        method_outargs = {}
        method_inargs = {}
        property_sig = {}
        for method in self.node_info.methods:
            method_outargs[method.name] = '(' + ''.join([arg.signature for arg in method.out_args]) + ')'
            method_inargs[method.name] = tuple(arg.signature for arg in method.in_args)
        self.method_inargs = method_inargs
        self.method_outargs = method_outargs

        self.con = Gio.bus_get_sync(Gio.BusType.SYSTEM, None)
        self.con.register_object(
            publish_path,
            self.node_info,
            self.handle_method_call,
            self.prop_getter,
            self.prop_setter)

    def handle_method_call(self,
                           connection: Gio.DBusConnection,
                           sender: str,
                           object_path: str,
                           interface_name: str,
                           method_name: str,
                           params: GLib.Variant,
                           invocation: Gio.DBusMethodInvocation
                           ):
        """
        This is the top-level function that handles method calls to
        the server.
        """
        args = list(params.unpack())
        # Handle the case where it is a Unix filedescriptor
        for i, sig in enumerate(self.method_inargs[method_name]):
            if sig == 'h':
                msg = invocation.get_message()
                fd_list = msg.get_unix_fd_list()
                args[i] = fd_list.get(args[i])
        func = self.__getattribute__(method_name)
[... 109 lines cut ...]
    try:
        mainloop.run()
    except KeyboardInterrupt:
        mainloop.quit()

if __name__ == '__main__':
    main()
```

This was a proof-of-concept to see if it was possible. I haven't done a lot of testing with it and I certainly make no claims that it is ready for production use or the required structure for use in the library.

This was the client I used to test it with:
```python
import socket

server_address = "xx:xx:xx:xx:xx:xx"
server_port = 1
with socket.socket(socket.AF_BLUETOOTH,
                   socket.SOCK_STREAM,
                   socket.BTPROTO_RFCOMM) as c:

    c.connect((server_address, server_port))
    c.send(b"desserts")
    print(c.recv(1024).decode())
```