pybluez/pybluez #408: Error building and installing in wondows 10 via pip

NervenCid, 2021-08-11:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

Hello i'm getting this error when i try to install pybluez via pip:.

 error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30037\\bin\\HostX86\\x64\\link.exe' failed with exit code 1181
  ----------------------------------------
  ERROR: Failed building wheel for pybluez
  Running setup.py clean for pybluez
Failed to build pybluez
Installing collected packages: pybluez
    Running setup.py install for pybluez ... error
    ERROR: Command errored out with exit status 1:

## System

 - **Operating System:** : Windows 10
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.) : PC
 - **Python Version:** : 3.9
 - **PyBluez Version:** : 0.23

## Issue

Please provide information necessary to reproduce the issue, including sample
code.

If you have long logs, please post them on https://gist.github.com and link to
the Gist here.

rayenmansouri, 2021-09-17:
try install libbluetooth-dev library for your windows machine

dogtopus, 2021-09-17:
Actually the CI still builds it fine. So please check your configuration or provide more detailed logs.

NervenCid, 2021-09-18:
@dogtopus thank you for your answer, this is the log when i try to install via pip

`WARNING: Ignoring invalid distribution -lask (c:\python39\lib\site-packages)
WARNING: Ignoring invalid distribution -ip (c:\python39\lib\site-packages)
WARNING: Ignoring invalid distribution -lask (c:\python39\lib\site-packages)
WARNING: Ignoring invalid distribution -ip (c:\python39\lib\site-packages)
Collecting pybluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
Building wheels for collected packages: pybluez
  Building wheel for pybluez (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\python39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\nerve\\AppData\\Local\\Temp\\pip-install-y60h0w7h\\pybluez_f5bd033e17fa4b45ab74c4e7d6ad4c83\\setup.py'"'"'; __file__='"'"'C:\\Users\\nerve\\AppData\\Local\\Temp\\pip-install-y60h0w7h\\pybluez_f5bd033e17fa4b45ab74c4e7d6ad4c83\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\nerve\AppData\Local\Temp\pip-wheel-i9e3ys30'
       cwd: C:\Users\nerve\AppData\Local\Temp\pip-install-y60h0w7h\pybluez_f5bd033e17fa4b45ab74c4e7d6ad4c83\
  Complete output (42 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.9
  creating build\lib.win-amd64-3.9\bluetooth
  copying bluetooth\ble.py -> build\lib.win-amd64-3.9\bluetooth
  copying bluetooth\bluez.py -> build\lib.win-amd64-3.9\bluetooth
  copying bluetooth\btcommon.py -> build\lib.win-amd64-3.9\bluetooth
  copying bluetooth\macos.py -> build\lib.win-amd64-3.9\bluetooth
  copying bluetooth\msbt.py -> build\lib.win-amd64-3.9\bluetooth
  copying bluetooth\widcomm.py -> build\lib.win-amd64-3.9\bluetooth
  copying bluetooth\__init__.py -> build\lib.win-amd64-3.9\bluetooth
  c:\python39\lib\site-packages\setuptools\lib2to3_ex.py:36: SetuptoolsDeprecationWarning: 2to3 support is deprecated. If the project still requires Python 2 support, please migrate to a single-codebase solution or employ an independent conversion process.
    warnings.warn(
  Fixing build\lib.win-amd64-3.9\bluetooth\ble.py build\lib.win-amd64-3.9\bluetooth\bluez.py build\lib.win-amd64-3.9\bluetooth\btcommon.py build\lib.win-amd64-3.9\bluetooth\macos.py build\lib.win-amd64-3.9\bluetooth\msbt.py build\lib.win-amd64-3.9\bluetooth\widcomm.py build\lib.win-amd64-3.9\bluetooth\__init__.py
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  Fixing build\lib.win-amd64-3.9\bluetooth\ble.py build\lib.win-amd64-3.9\bluetooth\bluez.py build\lib.win-amd64-3.9\bluetooth\btcommon.py build\lib.win-amd64-3.9\bluetooth\macos.py build\lib.win-amd64-3.9\bluetooth\msbt.py build\lib.win-amd64-3.9\bluetooth\widcomm.py build\lib.win-amd64-3.9\bluetooth\__init__.py
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  running build_ext
  building 'bluetooth._msbt' extension
  creating build\temp.win-amd64-3.9
  creating build\temp.win-amd64-3.9\Release
  creating build\temp.win-amd64-3.9\Release\msbt
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I.\port3 -Ic:\python39\include -Ic:\python39\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\ATLMFC\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcmsbt\_msbt.c /Fobuild\temp.win-amd64-3.9\Release\msbt\_msbt.obj
  _msbt.c
  msbt\_msbt.c(145): warning C4244: '=': conversi¢n de 'SOCKET' a 'int'; posible p‚rdida de datos
  msbt\_msbt.c(222): warning C4244: '=': conversi¢n de 'SOCKET' a 'int'; posible p‚rdida de datos
  msbt\_msbt.c(402): warning C4244: '=': conversi¢n de 'SOCKET' a 'int'; posible p‚rdida de datos
  msbt\_msbt.c(415): warning C4244: 'inicializando': conversi¢n de 'SOCKET' a 'int'; posible p‚rdida de datos
  msbt\_msbt.c(480): warning C4996: 'PyUnicode_FromUnicode': deprecated in 3.3
  msbt\_msbt.c(635): warning C4244: 'inicializando': conversi¢n de 'SOCKET' a 'int'; posible p‚rdida de datos
  msbt\_msbt.c(807): warning C4311: 'conversi¢n de tipo': truncamiento de puntero de 'HANDLE' a 'unsigned long'
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:c:\python39\libs /LIBPATH:c:\python39\PCbuild\amd64 /LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\ATLMFC\lib\x64 /LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\lib\x64 /LIBPATH:C:\Program Fi [... cut ...] ve\AppData\Local\Temp\pip-record-hsr_d_y2\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\python39\Include\pybluez' Check the logs for full command output.
WARNING: Ignoring invalid distribution -lask (c:\python39\lib\site-packages)
WARNING: Ignoring invalid distribution -ip (c:\python39\lib\site-packages)
WARNING: Ignoring invalid distribution -lask (c:\python39\lib\site-packages)
WARNING: Ignoring invalid distribution -ip (c:\python39\lib\site-packages)`

yysxzf, 2021-09-18:
In windows 10, irprops.lib has been removed from the Windows SDK. Apps that were linking against irprops.lib can switch to bthprops.lib as a drop-in replacement. First,make sure you have installed win10 SDK,then you can do like this.
··· 
cd C:\Program Files (x86)\Windows Kits\10\Lib\10.0.19041.0\um\x64  # Maybe you have different version of SDK.
mklink irprops bthprops.lib # run as system administrator 
pip install pybluez
···

dogtopus, 2021-09-18:
Oh yeah, the stable version is actually broken for quite a while. I believe the irprops dependency has been dropped since long ago. Meanwhile you can install the dev version from git directly.

closed 2021-09-18 by dogtopus

GreyBlackSKY, 2021-11-25:
> In windows 10, irprops.lib has been removed from the Windows SDK. Apps that were linking against irprops.lib can switch to bthprops.lib as a drop-in replacement. First,make sure you have installed win10 SDK,then you can do like this. ··· cd C:\Program Files (x86)\Windows Kits\10\Lib\10.0.19041.0\um\x64 # Maybe you have different version of SDK. mklink irprops.lib bthprops.lib # run as system administrator pip install pybluez ···

worked for me. thank you so very much. I was trying so hard and couldn't find the problem. you're an absolute legend.

BeanKorea, 2023-07-04:
> In windows 10, irprops.lib has been removed from the Windows SDK. Apps that were linking against irprops.lib can switch to bthprops.lib as a drop-in replacement. First,make sure you have installed [...]

this should be in the docs