pybluez/pybluez #409: "bt.connect (self._sockfd, addr, port) OSError"

lin799891, 2021-08-19:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:win10
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:3.6.8
 - **PyBluez Version:0.22

## Issue

''bt.connect (self._sockfd, addr, port)
OSError''

What I want to know is whether there is an effective solution to this problem now？

Kpakizeh, 2021-08-21:
I have the same problem and I was not able to fix it, yet. Please let me know if you found a way. I will do the same. 
I saw [this ](https://github.com/pybluez/pybluez/issues/349) and tried with  pyBluez 0.22 but no luck.

dogtopus, 2021-08-23:
The latest stable version on PyPI is very outdated. Please install PyBluez from git directly.

closed 2021-09-16 by dogtopus