pybluez/pybluez #411: maintainers

hiaselhans, 2021-09-08:
Hello,

It seems this library has finally gotten a new maintainer..?

I was a little upset about the state of pybluez some time ago so i started my own fork:
https://github.com/airgproducts/pybluez2

i did a few things there:

- automatic deployment to pypi with gh-actions
- remove widcomm
- cleanup folders
- moved to pure pyobjc (remove aquablue/lightaquablue) in osx because deployment was a little nightmare...

I'd be happy to contribute any of those to pybluez in the form of PR's but would like to know if you are interested first :)

dogtopus, 2021-09-08:
>     * moved to pure pyobjc (remove aquablue/lightaquablue) in osx because deployment was a little nightmare...

This is great. I am however unable to test whether it works or not, but I guess it's probably fine to merge it since lightaquablue is already rotten to the point of pretty much unusable anyway.

hiaselhans, 2021-09-08:
> This is great. I am however unable to test whether it works or not, but I guess it's probably fine to merge it since lightaquablue is already rotten to the point of pretty much unusable anyway.

I am using disciver_devices and the serial port in production. However, i didnt care about the rest of pybluez‘ functions and we might have to see if any other parts of it are used? For me discovery with mac addresses and serial port was what i needed.

dogtopus, 2021-11-05:
I think that's pretty much what is offered across all 3 platforms anyway. I'm not even sure whether low level L2CAP sockets are ever supported by PyBluez on macOS.

dogtopus, 2021-12-06:
@hiaselhans could you please isolate the macos part of your fork as a PR so I could merge it? Pretty sure anything is better than the current broken macos port.

hiaselhans, 2021-12-07:
Hi @dogtopus 

I can do that, but it would take me some time.

Let's say i can do it when we can get automatic deployment to pypi in as well?
I think thats a major issue with pybluez we have to address as well and i don't want to waste my time ;)

dogtopus, 2021-12-07:
@hiaselhans we probably need to ask the main maintainer for pypi credentials first since I don't think I have pypi access.

So maybe later for the pypi auto deploy?

Also @rgov what do you think?

rgov, 2021-12-07:
I am not particularly active here but I would merge in @hiaselhans's changes and help you get access to the PyPI account. I'm not sure automatic deployment is a great plan since we have no way to run automated tests.

hiaselhans, 2021-12-08:
on the other hand, doing manual deployments on microsoft, linux and macos seems like a task i wouldn't ever do manually....?

rgov, 2021-12-08:
We have continuous integration https://github.com/pybluez/pybluez/blob/master/.github/workflows/build.yml that can produce built packages for each operating system, and then manually release these artifacts to PyPI.

hiaselhans, 2021-12-08:
so, the plan is to have wheels built by gh-actions for python (3.?-3.10) x (win, osx, linux) and upload those 12 wheel-files + source dist to pypi manually?

rgov, 2021-12-08:
Yes, but we will need to change the CI system to export the artifacts for download, though.

hiaselhans, 2021-12-10:
I disagree with that, because I see so many packages on github where it is an endless road to get the (not-so-active) maintainers to publish a release.

In my point of view not having an automated release process will once again leave this package in a state where close to nobody is able to install it.
I mean look where we are now: python3.8 is out since 2 years now and the maximum supported version on pypi is 3.7

schodet, 2021-12-10:
If I understand correctly, @hiaselhans has done the work on its branch to automate the releases to PyPI, so it’s “almost” there. If testing is the barrier, maybe just change it to only make a release when there is a tag, so that the master branch can be tested and if everything goes well, tag and release.

Do we have testers on all three platforms?

closed 2022-04-27 by stale[bot]