pybluez/pybluez #412: PyBluez doenst respond or finish anything

Florian7T, 2021-09-10:
## System

 - **Operating System: Kali GNU/Linux Rolling x86_64 
 - **Hardware: Realtek RTL8822CE
 - **Python Version: 3.9
 - **PyBluez Version: 0.23

PyBluez doenst respond or finish anything. Running the code below results in just waiting for anything to happen but after a good 3 minutes I gave up. I've tried playing with arguments such as lookup_names and duration but nothing changed... Am I doing something wrong? The code doesn't exit either.

`import bluetooth
bluetooth.discover_devices()`

Florian7T, 2021-09-10:
it somehow does work on Win10? It's no problem I dual boot anyways :) It's probably a fault on my side because no one else is has my problem

closed 2021-09-10 by Florian7T

dogtopus, 2021-09-10:
The Windows backend is the one that ironically receieving the most fixes recently. There's a fork that addresses some issues on Mac by rewriting the backend code to not use LightAquaBlue. The next will be Linux once we figure out what to do with the dbus Bluez5 API.

Florian7T, 2021-09-12:
> The Windows backend is the one that ironically receieving the most fixes recently. There's a fork that addresses some issues on Mac by rewriting the backend code to not use LightAquaBlue. The next will be Linux once we figure out what to do with the dbus Bluez5 API.

Unlucky im trying so hard to learn things on linux (hence the dual boot) but things like this just keep me from using it 😅