pybluez/pybluez #416: Planned releases on Pypi ?

Tropicao, 2021-09-28:
Hi,
I rely on Pybluez for some project. I have stumbled upon the [`2to3` issue](https://github.com/pybluez/pybluez/issues/413), but it seems to be fixed on mainline since a few days :tada: 
I am used to pull pybluez through pip, so I would be glad to get this fix through a new version on Pypi, but I see there have been no release since end of 2019 on Pypi, nor tags on the repository.
Is there any new release planned ? Or is there any pending major issue,  preventing a version publication ?
May be linked to https://github.com/pybluez/pybluez/issues/411

jantman, 2022-07-13:
Any hope of either getting a new release out, or adding someone else as a maintainer/owner who can handle that?

Mimoja, 2022-08-02:
Same painpoint from ubuntu 22.04 perspecive shipping with setuptools 59.6.0 here

dvzrv, 2023-04-07:
Hi! we're currently doing rebuilds for Python 3.11 on Arch Linux and the last release 0.23 (from 2019!) does not work with it.

Please create a new release or add someone to the project to allow this issue to move forward. Thanks! :)

l3d00m, 2023-07-05:
Hi @dogtopus. I am aware that you no longer actively maintain the module. Since you still accept pull requests, it would be great if you could do another release to pypi. 

That way people aren't forced to directly build from master. It's becoming more of a problem now that python 3.11 adoption is picking up.

OrangeDog, 2023-10-16:
The fork https://github.com/airgproducts/pybluez2 is published on PyPi, which people may find a useful alternative.

blshkv, 2024-06-06:
> The fork https://github.com/airgproducts/pybluez2 is published on PyPi, which people may find a useful alternative.
> 
> There are also `pybluez-edge`, `PyBluez-updated`, and `PyBluezovenlab`.

that fork seem dead too:
```71 commits ahead, 12 commits ```
And no releases since 2022

j-romchain, 2024-06-19:
By the way, the solution is to add a workflow to push it to pypi with each accepted pull request or somesuch. that way they don't have to do anything beyond accepting pull requests.

prusswan, 2024-06-27:
[PyBluez-0.30dev-cp37-cp37m-win_amd64.whl.zip](https://github.com/user-attachments/files/16020451/PyBluez-0.30dev-cp37-cp37m-win_amd64.whl.zip)
Built against master recently for Python 3.7

OrangeDog, 2025-04-09:
`pybluez-edge` now also fails to build:

```
#9 4.115 Building wheels for collected packages: pybluez-edge
#9 4.116   Building wheel for pybluez-edge (setup.py): started
#9 4.384   Building wheel for pybluez-edge (setup.py): finished with status 'error'
#9 4.391   error: subprocess-exited-with-error
#9 4.391   
#9 4.391   �� python setup.py bdist_wheel did not run successfully.
#9 4.391   ��� exit code: 1
#9 4.391   ������> [21 lines of output]
#9 4.391       Traceback (most recent call last):
#9 4.391         File "<string>", line 2, in <module>
#9 4.391         File "<pip-setuptools-caller>", line 34, in <module>
#9 4.391         File "/tmp/pip-install-9f1ahag7/pybluez-edge_d9533bfb0b3647a3b8acb73c62fb07d9/setup.py", line 84, in <module>
#9 4.391           setup(name='pybluez-edge',
#9 4.391         File "/usr/local/lib/python3.11/site-packages/setuptools/__init__.py", line 108, in setup
#9 4.391           return distutils.core.setup(**attrs)
#9 4.391                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
#9 4.391         File "/usr/local/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 172, in setup
#9 4.391           ok = dist.parse_command_line()
#9 4.391                ^^^^^^^^^^^^^^^^^^^^^^^^^
#9 4.391         File "/usr/local/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 475, in parse_command_line
#9 4.391           args = self._parse_command_opts(parser, args)
#9 4.391                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
#9 4.391         File "/usr/local/lib/python3.11/site-packages/setuptools/dist.py", line 1111, in _parse_command_opts
#9 4.391           nargs = _Distribution._parse_command_opts(self, parser, args)
#9 4.391                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
#9 4.391         File "/usr/local/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 540, in _parse_command_opts
#9 4.391           if not issubclass(cmd_class, Command):
#9 4.391                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
#9 4.391       TypeError: issubclass() arg 1 must be a class
#9 4.391       [end of output]
```