pybluez/pybluez #418: L2CAP does not work

KillerBOSS2019, 2021-10-07:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System: Windows 10 64bit** 
 - **Hardware:  External Bluetooth Adaptor** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version: 3.9.6**
 - **PyBluez Version: pybluez-0.30-py3.9-win-amd64.egg**

## Issue

I can't seems able to figure out how to connect to a L2CAP bluetooth device it always give me error says "ValueError: invalid protocol"
This is my code
```py
remote = bluetooth.find_service(address=macAddress)[0]
# print(remote)
# print(bluetooth.L2CAP)
s = bluetooth.BluetoothSocket(bluetooth.L2CAP)
s.connect((remote['host'], remote['port']))
s.recv(4000)
```
also on top of that i tried to `print(bluetooth.L2CAP)` and it gave me value `0`

KillerBOSS2019, 2021-10-08:
wait at https://docs.microsoft.com/en-us/windows-hardware/drivers/bluetooth/bluetooth-driver-stack it says 
```
RfComm- The component that implements the Bluetooth serial cable emulation protocol. RfComm also uses the L2CAP and SDP interfaces found in BthPort. The upper edge of RfComm exposes the TDI interface, allowing this component to appear to be a networking transport. This is how WshBth connects to Bluetooth to send and receive data from user-mode APIs.
```
it says RFCOMM uses L2CAP. also a question If the L2CAP wasnt supported then how is this L2CAP BT device connect to this Windows?

dogtopus, 2021-10-08:
> it says RFCOMM uses L2CAP.

Yes. However there's no low level L2CAP support. Only RFCOMM.

closed 2021-10-31 by KillerBOSS2019