pybluez/pybluez #421: The following error was encountered while installing pybluez on macmini2020 (M1). I really hope to get your help

1997shp, 2021-11-02:
ERROR: Command errored out with exit status 1:
   command: /opt/homebrew/opt/python@3.9/bin/python3.9 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/jv/rbxjnrd953qbvf105vr9gysm0000gn/T/pip-install-aelp84ol/pybluez_f0e9dced9a7949c199843dbdab186e51/setup.py'"'"'; __file__='"'"'/private/var/folders/jv/rbxjnrd953qbvf105vr9gysm0000gn/T/pip-install-aelp84ol/pybluez_f0e9dced9a7949c199843dbdab186e51/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/jv/rbxjnrd953qbvf105vr9gysm0000gn/T/pip-wheel-_3zdkh9d
       cwd: /private/var/folders/jv/rbxjnrd953qbvf105vr9gysm0000gn/T/pip-install-aelp84ol/pybluez_f0e9dced9a7949c199843dbdab186e51/
  Complete output (21 lines):
  objc[18039]: Class AMSupportURLConnectionDelegate is implemented in both ?? (0x1f81738f0) and ?? (0x115fb02b8). One of the two will be used. Which one is undefined.
  objc[18039]: Class AMSupportURLSession is implemented in both ?? (0x1f8173940) and ?? (0x115fb0308). One of the two will be used. Which one is undefined.
  Command line invocation:
      /Applications/Xcode.app/Contents/Developer/usr/bin/xcodebuild install -project macos/LightAquaBlue/LightAquaBlue.xcodeproj -scheme LightAquaBlue DSTROOT=/private/var/folders/jv/rbxjnrd953qbvf105vr9gysm0000gn/T/pip-install-aelp84ol/pybluez_f0e9dced9a7949c199843dbdab186e51/macos INSTALL_PATH=/ DEPLOYMENT_LOCATION=YES
  
  User defaults from command line:
      IDEPackageSupportUseBuiltinSCM = YES
  
  Build settings from command line:
      DEPLOYMENT_LOCATION = YES
      DSTROOT = /private/var/folders/jv/rbxjnrd953qbvf105vr9gysm0000gn/T/pip-install-aelp84ol/pybluez_f0e9dced9a7949c199843dbdab186e51/macos
      INSTALL_PATH = /
  
  xcodebuild: error: 'macos/LightAquaBlue/LightAquaBlue.xcodeproj' does not exist.
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/private/var/folders/jv/rbxjnrd953qbvf105vr9gysm0000gn/T/pip-install-aelp84ol/pybluez_f0e9dced9a7949c199843dbdab186e51/setup.py", line 79, in <module>
      subprocess.check_call([
    File "/opt/homebrew/Cellar/python@3.9/3.9.2_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['xcodebuild', 'install', '-project', 'macos/LightAquaBlue/LightAquaBlue.xcodeproj', '-scheme', 'LightAquaBlue', 'DSTROOT=/private/var/folders/jv/rbxjnrd953qbvf105vr9gysm0000gn/T/pip-install-aelp84ol/pybluez_f0e9dced9a7949c199843dbdab186e51/macos', 'INSTALL_PATH=/', 'DEPLOYMENT_LOCATION=YES']' returned non-zero exit status 66.
  ----------------------------------------
  ERROR: Failed building wheel for pybluez

hiwanz, 2021-11-05:
Some issue with python@3.9

dogtopus, 2021-11-05:
Maybe give this [fork](https://github.com/airgproducts/pybluez2) a try? PyBluez's macOS support is broken since quite a while ago.

N8866U, 2021-11-06:
I have the same issue on an Ubuntu 20.04 machine.

Mohmmde1, 2021-11-18:
same issue here as well, I have tried using pip but didn't work, then I cloned this repo, I tried run the setup file however it didn't work

closed 2022-04-27 by stale[bot]