pybluez/pybluez #424: couldn't install the packge

Mohmmde1, 2021-11-18:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - macOS Big Sur version 11.5.2
 - MacBook Air (Retina, 13-inch, 2020) 
 - 3.9.5
 -  couldn't download the last version

## Issue
This is the message that I received when I run this line in my virtual env pip install pybluez

Collecting pybluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Preparing metadata (setup.py) ... error
  **ERROR: Command errored out with exit status 1:
   command: /Users/mohammedalnashrei/PycharmProjects/connect/venv/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/hk/6378r6kn485c937c46zgdhy80000gn/T/pip-install-ng8la3zd/pybluez_055f298e4b934d0da0d8e2197099a4d1/setup.py'"'"'; __file__='"'"'/private/var/folders/hk/6378r6kn485c937c46zgdhy80000gn/T/pip-install-ng8la3zd/pybluez_055f298e4b934d0da0d8e2197099a4d1/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/hk/6378r6kn485c937c46zgdhy80000gn/T/pip-pip-egg-info-oia270qg
       cwd: /private/var/folders/hk/6378r6kn485c937c46zgdhy80000gn/T/pip-install-ng8la3zd/pybluez_055f298e4b934d0da0d8e2197099a4d1/
  Complete output (1 lines):
  error in PyBluez setup command: use_2to3 is invalid.
  ----------------------------------------**
WARNING: Discarding https://files.pythonhosted.org/packages/08/9f/e9d93b266d2d1ea988780a52a696073ba0a65df65a532165fdf6ff90d0ed/PyBluez-0.23.tar.gz#sha256=c8f04d2e78951eaa9de486b4d49381704e8943d0a6e6e58f55fcd7b8582e90de (from https://pypi.org/simple/pybluez/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Downloading PyBluez-0.22.zip (109 kB)
     |████████████████████████████████| 109 kB 3.6 MB/s            
  Preparing metadata (setup.py) ... done
Using legacy 'setup.py install' for pybluez, since package 'wheel' is not installed.
Installing collected packages: pybluez
    Running setup.py install for pybluez ... error
    ERROR: Command errored out with exit status 1:
     command: /Users/mohammedalnashrei/PycharmProjects/connect/venv/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/hk/6378r6kn485c937c46zgdhy80000gn/T/pip-install-ng8la3zd/pybluez_af88741f88194f17b6356b76f460bd83/setup.py'"'"'; __file__='"'"'/private/var/folders/hk/6378r6kn485c937c46zgdhy80000gn/T/pip-install-ng8la3zd/pybluez_af88741f88194f17b6356b76f460bd83/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/hk/6378r6kn485c937c46zgdhy80000gn/T/pip-record-ds71xzg4/install-record.txt --single-version-externally-managed --compile --install-headers /Users/mohammedalnashrei/PycharmProjects/connect/venv/include/site/python3.9/pybluez
         cwd: /private/var/folders/hk/6378r6kn485c937c46zgdhy80000gn/T/pip-install-ng8la3zd/pybluez_af88741f88194f17b6356b76f460bd83/
    Complete output (61 lines):
    running install
    /Users/mohammedalnashrei/PycharmProjects/connect/venv/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
      warnings.warn(
    running build
    running build_py
    creating build
    creating build/lib.macosx-10.9-x86_64-3.9
    creating build/lib.macosx-10.9-x86_64-3.9/bluetooth
    copying bluetooth/btcommon.py -> build/lib.macosx-10.9-x86_64-3.9/bluetooth
    copying bluetooth/ble.py -> build/lib.macosx-10.9-x86_64-3.9/bluetooth
    copying bluetooth/__init__.py -> build/lib.macosx-10.9-x86_64-3.9/bluetooth
    copying bluetooth/bluez.py -> build/lib.macosx-10.9-x86_64-3.9/bluetooth
    copying bluetooth/msbt.py -> build/lib.macosx-10.9-x86_64-3.9/bluetooth
    copying bluetooth/widcomm.py -> build/lib.macosx-10.9-x86_64-3.9/bluetooth
    copying bluetooth/osx.py -> build/lib.macosx-10.9-x86_64-3.9/bluetooth
    running build_ext
    building 'bluetooth._osxbt' extension
    creating build/temp.macosx-10.9-x86_64-3.9
    creating build/temp.macosx-10.9-x86_64-3.9/osx
    gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch x86_64 -g -I/System/Library/Frameworks/IOBluetooth.framework/Headers -I/System/Library/Frameworks/CoreFoundation.framework/Headers -I/Users/mohammedalnashrei/PycharmProjects/connect/venv/include -I/Library/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c osx/_osxbt.c -o build/temp.macosx-10.9-x86_64-3.9/osx/_osxbt.o
    osx/_osxbt.c:676:5: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
        IOBluetoothDeviceInquiryRef inquiry;
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
        IOBluetoothDeviceRef
    /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOBluetooth.framework/Headers/IOBluetoothUserLib.h:34:47: note: 'IOBluetoothDeviceRef' declared here
    typedef struct OpaqueIOBluetoothObjectRef *                     IOBluetoothDeviceRef;
                                                                    ^
    osx/_osxbt.c:688:17: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
                    IOBluetoothDeviceInquiryRef inquiryRef,
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                    IOBluetoothDeviceRef
    /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOBluetooth.framework/Headers/IOBluetoothUserLib.h:34:47: note: 'IOBluetoothDeviceRef' declared here
    typedef struct OpaqueIOBluetoothObjectRef *                     IOBluetoothDeviceRef;
                                                                    ^
    osx/_osxbt.c:703:19: error: implicit declaration of function 'IOBluetoothDeviceInquiryCreateWithCallbackRefCon' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        dd->inquiry = IOBluetoothDeviceInquiryCreateWithCallbackRefCon (&dd);
                      ^
[... 1 lines cut ...]
        dd->inquiry = IOBluetoothDeviceInquiryCreateWithCallbackRefCon (&dd);
                    ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    osx/_osxbt.c:705:5: error: implicit declaration of function 'IOBluetoothDeviceInquirySetCompleteCallback' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        IOBluetoothDeviceInquirySetCompleteCallback (dd->inquiry,
        ^
    osx/_osxbt.c:717:5: error: implicit declaration of function 'IOBluetoothDeviceInquiryStart' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        IOBluetoothDeviceInquiryStart (dd->inquiry);
        ^
    osx/_osxbt.c:721:5: error: implicit declaration of function 'IOBluetoothDeviceInquiryDelete' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        IOBluetoothDeviceInquiryDelete (dd->inquiry);
        ^
    osx/_osxbt.c:721:5: note: did you mean 'IOBluetoothDeviceInquiryStart'?
    osx/_osxbt.c:717:5: note: 'IOBluetoothDeviceInquiryStart' declared here
        IOBluetoothDeviceInquiryStart (dd->inquiry);
        ^
    osx/_osxbt.c:774:9: error: implicit declaration of function 'Py_InitModule3' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        m = Py_InitModule3("_osxbt", osxbt_methods, osxbt_doc);
            ^
    osx/_osxbt.c:774:7: warning: incompatible integer to pointer conversion assigning to 'PyObject *' (aka 'struct _object *') from 'int' [-Wint-conversion]
        m = Py_InitModule3("_osxbt", osxbt_methods, osxbt_doc);
          ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    2 warnings and 7 errors generated.
    error: command '/usr/bin/gcc' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /Users/mohammedalnashrei/PycharmProjects/connect/venv/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/hk/6378r6kn485c937c46zgdhy80000gn/T/pip-install-ng8la3zd/pybluez_af88741f88194f17b6356b76f460bd83/setup.py'"'"'; __file [... cut ...] path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/hk/6378r6kn485c937c46zgdhy80000gn/T/pip-record-ds71xzg4/install-record.txt --single-version-externally-managed --compile --install-headers /Users/mohammedalnashrei/PycharmProjects/connect/venv/include/site/python3.9/pybluez Check the logs for full command output.

SpudGunMan, 2022-01-29:
I am having similar issues of non install on new Debian install as well due to missing dependency files

SpudGunMan, 2022-01-31:
I think osx12.3 may have a fix for this

miigotu, 2022-02-25:
https://github.com/pybluez/pybluez/issues/433#issuecomment-1050613900