pybluez/pybluez #425: Any alternative to this?

hkayann, 2021-11-23:
This package seems like the ones that will eventually will disappear into the void.

Is there any alternative to this as I would like to escape ugly D-Bus API? 

I can contribute, if there is any alternative which is under active development.

dogtopus, 2021-11-25:
If you only want BLE there are a bunch of alternatives I think. Otherwise I didn't really know any libraries that simplifies the DBus stuff.

If you want to go for the DBus route maybe take a look at nxbt's Bluetooth support code for some inspirations.

SujithChristopher, 2022-01-08:
I think this library is better @hkayann 

https://github.com/hbldh/bleak

dogtopus, 2022-01-08:
> I think this library is better @hkayann
> 
> https://github.com/hbldh/bleak

Pretty sure that's BLE only and PyBluez isn't just about BLE.

closed 2022-04-27 by stale[bot]