pybluez/pybluez #426: pybluez with python 3.10 requires PY_SSIZE_T_CLEAN macro in # format

rudyvan, 2021-11-29:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - Debian bullseye (11)
 - Raspberry Pi, external Bluetooth adaptor
 - python 3.10.0
 - pybluez version: latest

## Issue

pybluez needs update for python 3.10 as then it requires PY_SSIZE_T_CLEAN macro in # format
From version 3.10 of python, the interface to C programs must be in this specific format

if you run below script, as of 3.10 you get a system error:

```
bluez.hci_send_cmd(self.btle_sock, OGF_LE_CTL, OCF_LE_SET_SCAN_ENABLE, cmd_pkt)
SystemError: PY_SSIZE_T_CLEAN macro must be defined for '#' formats
```

closed 2022-01-03 by dogtopus

axilirator, 2022-02-13:
Can we please have a patch release containing this fix? Thanks!

kred-jon, 2022-04-26:
SystemError: PY_SSIZE_T_CLEAN macro must be defined for '#' formats

Python 3.10.4

Ubuntu 22.04 LTS

commit cf6217c3 (2022-01-03): Use Py_ssize_t when parsing buffer length, fix #426 (#427)

From python 3.9 documentation:

> For all # variants of formats (s#, y#, etc.), the macro
> PY_SSIZE_T_CLEAN must be defined before including Python.h. On Python
> 3.9 and older, the type of the length argument is Py_ssize_t if the
> PY_SSIZE_T_CLEAN macro is defined, or int otherwise.

From python 3.8 changes:

> Use of # variants of formats in parsing or building value (e.g.
> PyArg_ParseTuple(), Py_BuildValue(), PyObject_CallFunction(), etc.)
> without PY_SSIZE_T_CLEAN defined raises DeprecationWarning now. It
> will be removed in  [... cut]