pybluez/pybluez #431: Cant install PyBluez on Windows: use_2to3 is invalid.

woolyplayer, 2022-02-03:
## System

 - **Windows 10 Home**
 - **latest (3.10)**
 - **pip and the zip**
 - **with the windows SDK**

## whenever i try to install pybluez via terminal and or Zip i get this 

```
Collecting PyBluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```
## note
I was able to recreate this error just by trying to install it on a VM 
also with the same  system
i tried it without installing the SDK didnt work and got lots of errors as (expectet) but i also got the error above
after installing the sdk i got just the error above

pls Help thx

dogtopus, 2022-02-03:
#413

closed 2022-02-03 by dogtopus

woolyplayer, 2022-02-04:
Ok so for anyone who wants to instal this:

- Dont use pip 
- make sure you have all dependencies installed
- in setup.py remove use_2to3=True on line 127
- change line 32 to: libraries=["WS2_32", "Bthprops"],

this might help

PRO-2684, 2022-04-24:
> Ok so for anyone who wants to instal this:
> 
> * Dont use pip
> * make sure you have all dependencies installed
> * in setup.py remove use_2to3=True on line 127
> * change line 32 to: libraries=["WS2_32", "Bthprops"],
> 
> this might help

Consider adding this to readme? It could help a lot.

dogtopus, 2022-04-24:
@PRO-2684 
The pypi version is outdated and this issue is fixed in the dev version ages ago. Please install it from source with `pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez`

It should be mentioned somewhere on the issue tracker but everyone seem to send duplicate reports without searching and eventually made searching more difficult.

Kronodile, 2022-06-11:
> @PRO-2684 The pypi version is outdated and this issue is fixed in the dev version ages ago. Please install it from source with `pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez` [...]

I get this error 
zsh: no matches found: https://github.com/pybluez/pybluez.git#egg=pybluez

Nassim-dev, 2022-11-27:
Thanks @dogtopus your line is working well 👌🏻

andburch, 2023-02-14:
Didn't work for me

baysahguwor, 2023-04-22:
can't work for me either

 pip install pybluez2
Collecting pybluez2
  Using cached pybluez2-0.46.tar.gz (50 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]

jamieduk, 2023-06-09:
incompetant fool didnt check hes work before releasing this is a total epic fail i expect better, im using windows 11 64 but python 3.11 tried multiple ways and it fails to install! fixit!!!

sluissantos, 2023-08-12:
Necessary Python 3.7 version

Download:
https://files.pythonhosted.org/packages/3e/78/d4ba3d69a20313be05afbea2ce5d29f7fa39f7a5c091fc7c1331a3feb3c1/PyBluez-0.23-cp37-cp37m-win_amd64.whl

pip install pybluez-0.23-cp37-cp37m-win_amd64.whl

pip install pyinstaller

obiscr, 2023-10-13:
> incompetant fool didnt check hes work before releasing this is a total epic fail i expect better, im using windows 11 64 but python 3.11 tried multiple ways and it fails to install! fixit!!!

I'm with you, win11 python 3.11, but I'm working.

[image]

WhiteTiger-y, 2024-03-16:
pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez
Collecting pybluez
  Cloning https://github.com/pybluez/pybluez.git to c:\users\yasha\appdata\local\temp\pip-install-_x0yx9sn\pybluez_09e8e925480244c1981b6df16f49d6ea
  Running command git clone --filter=blob:none --quiet https://github.com/pybluez/pybluez.git 'C:\Users\yasha\AppData\Local\Temp\pip-install-_x0yx9sn\pybluez_09e8e925480244c1981b6df16f49d6ea'
  Resolved https://github.com/pybluez/pybluez.git to commit 82cbba8a1ebd4c1e3442dfafd8581d58c50fa39e
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: pybluez
  Building wheel for pybluez (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for pybluez (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [15 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build\lib.win-amd64-cpython-312
      creating build\lib.win-amd64-cpython-312\bluetooth
      copying bluetooth\ble.py -> build\lib.win-amd64-cpython-312\bluetooth
      copying bluetooth\bluez.py -> build\lib.win-amd64-cpython-312\bluetooth
      copying bluetooth\btcommon.py -> build\lib.win-amd64-cpython-312\bluetooth
      copying bluetooth\macos.py -> build\lib.win-amd64-cpython-312\bluetooth
      copying bluetooth\msbt.py -> build\lib.win-amd64-cpython-312\bluetooth
      copying bluetooth\__init__.py -> build\lib.win-amd64-cpython-312\bluetooth
      running build_ext
      building 'bluetooth._msbt' extension
      error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pybluez
Failed to build pybluez
ERROR: Could not build wheels for pybluez, which is required to install pyproject.toml-based projects

seanlim4, 2024-03-20:
@WhiteTiger-y perhaps you should get the ms c++ build tools...

DUK2K, 2024-05-06:
> @WhiteTiger-y perhaps you should get the ms c++ build tools...

I have the same problem with the comment above. I already have the version 14 or above but it still gives me the same error.
[image]

Mausy5043, 2024-06-09:
The fix provided by @dogtopus nolonger works.
```
$ pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez
Defaulting to user installation because normal site-packages is not writeable
Collecting pybluez
  Cloning https://github.com/pybluez/pybluez.git to /tmp/pip-install-e76gy1xk/pybluez_dab7c10d5dda442baad901affa2fcb11
  Running command git clone --filter=blob:none --quiet https://github.com/pybluez/pybluez.git /tmp/pip-install-e76gy1xk/pybluez_dab7c10d5dda442baad901affa2fcb11
  Resolved https://github.com/pybluez/pybluez.git to commit 82cbba8a1ebd4c1e3442dfafd8581d58c50fa39e
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: pybluez
  Building wheel for pybluez (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [23 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/msbt.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/macos.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/btcommon.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/bluez.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/ble.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/__init__.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      running build_ext
      building 'bluetooth._bluetooth' extension
      creating build/temp.linux-aarch64-cpython-311
      creating build/temp.linux-aarch64-cpython-311/bluez
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/include/python3.11 -c bluez/btmodule.c -o build/temp.linux-aarch64-cpython-311/bluez/btmodule.o
      In file included from bluez/btmodule.c:21:
      bluez/btmodule.h:5:10: fatal error: bluetooth/bluetooth.h: No such file or directory
          5 | #include <bluetooth/bluetooth.h>
            |          ^~~~~~~~~~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pybluez
  Running setup.py clean for pybluez
Failed to build pybluez
Installing collected packages: pybluez
  Running setup.py install for pybluez ... error
  error: subprocess-exited-with-error

  × Running setup.py install for pybluez did not run successfully.
  │ exit code: 1
  ╰─> [25 lines of output]
      running install
      /usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/msbt.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/macos.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/btcommon.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/bluez.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/ble.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      copying bluetooth/__init__.py -> build/lib.linux-aarch64-cpython-311/bluetooth
      running build_ext
      building 'bluetooth._bluetooth' extension
      creating build/temp.linux-aarch64-cpython-311
      creating build/temp.linux-aarch64-cpython-311/bluez
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/include/python3.11 -c bluez/btmodule.c -o build/temp.linux-aarch64-cpython-311/bluez/btmodule.o
      In file included from bluez/btmodule.c:21:
      bluez/btmodule.h:5:10: fatal error: bluetooth/bluetooth.h: No such file or directory
          5 | #include <bluetooth/bluetooth.h>
            |          ^~~~~~~~~~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pybluez

note: This is an issue with the package mentioned above, not pip.
```

Omarch47, 2024-06-14:
> The fix provided by @dogtopus nolonger works.
>
> ```
> $ pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez
> Defaulting to user installation because normal site-packages is not writeable [...]

I was able to fix this by running `sudo apt install libbluetooth-dev
`

harryhanYuhao, 2024-06-22:
Can we change the `docs/install.rst` to reflect this? 

I don't think `pip install pybluez` would work at the moment at all for any OS. 

Or can we update the pip index?

Yembot31013, 2024-06-26:
I tried `pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez` but I got this error below

```
(env) 127_0_0_1_is_safer@127:~/cyber/BlueDucky$ pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez
Collecting pybluez
  Cloning https://github.com/pybluez/pybluez.git to /tmp/pip-install-do0qwe2n/pybluez_e752d70a666340cd9aaf0340600677fb
  Running command git clone --filter=blob:none --quiet https://github.com/pybluez/pybluez.git /tmp/pip-install-do0qwe2n/pybluez_e752d70a666340cd9aaf0340600677fb
  Resolved https://github.com/pybluez/pybluez.git to commit 82cbba8a1ebd4c1e3442dfafd8581d58c50fa39e
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [58 lines of output]
      running dist_info
      creating /tmp/pip-modern-metadata-mg37iuob/PyBluez.egg-info
      writing /tmp/pip-modern-metadata-mg37iuob/PyBluez.egg-info/PKG-INFO
      writing dependency_links to /tmp/pip-modern-metadata-mg37iuob/PyBluez.egg-info/dependency_links.txt
      writing eager_resources to /tmp/pip-modern-metadata-mg37iuob/PyBluez.egg-info/eager_resources.txt
      writing requirements to /tmp/pip-modern-metadata-mg37iuob/PyBluez.egg-info/requires.txt
      writing top-level names to /tmp/pip-modern-metadata-mg37iuob/PyBluez.egg-info/top_level.txt
      writing manifest file '/tmp/pip-modern-metadata-mg37iuob/PyBluez.egg-info/SOURCES.txt'
      reading manifest file '/tmp/pip-modern-metadata-mg37iuob/PyBluez.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      warning: no files found matching 'README'
      warning: no files found matching '*.c' under directory 'macos'
      warning: no previously-included files matching '*.pyc' found under directory '*'
      warning: no previously-included files matching '*.o' found under directory '*'
      warning: no previously-included files matching '*.opp' found under directory '*'
      adding license file 'COPYING'
      writing manifest file '/tmp/pip-modern-metadata-mg37iuob/PyBluez.egg-info/SOURCES.txt'
      creating '/tmp/pip-modern-metadata-mg37iuob/PyBluez-0.30.dist-info'
      Traceback (most recent call last):
        File "/home/codewithyembot/cyber/BlueDucky/env/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/codewithyembot/cyber/BlueDucky/env/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/codewithyembot/cyber/BlueDucky/env/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 149, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 368, in prepare_metadata_for_build_wheel
          self.run_setup()
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 497, in run_setup
          super().run_setup(setup_script=setup_script)
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 313, in run_setup
          exec(code, locals())
        File "<string>", line 84, in <module>
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 184, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 200, in run_commands
          dist.run_commands()
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/dist.py", line 976, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/command/dist_info.py", line 97, in run
          bdist_wheel = self.get_finalized_command('bdist_wheel')
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-o0c9qsrf/overlay/lib/python3.12/site-packages/setuptools/_distutils/cmd.py", line 302, in get_finalized_command
          cmd_obj = self.distribution.get_command_obj(command, create)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 11 lines cut ...]

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

This are what I used to solve it and it work fine for me:

[... cut ...]
   pip install --upgrade pip setuptools wheel
   ```

2. **I then manually Install the Build Dependencies before trying to install `pybluez`**:
   ```bash
   pip install cython
   ```

3. **Instead of installing directly from the URL, Then this time I try Cloning the Repository and Installing from the Local directory**:
   ```bash
   git clone https://github.com/pybluez/pybluez.git
   cd pybluez
   python setup.py install
   ```

Those are the step I took to resolve the issue. I am using ubuntu directly from laptop (not from VM)

madopal, 2024-07-07:
BTW @Yembot31013 's solution works within pip installing from repository as well. Don't need step 3. Seems like the setuptools/wheel/cython combo are what fixes it.