pybluez/pybluez #433: Can't install

sergoum, 2022-02-22:
I am trying to install the package but pip keeps failing. The error I get is this:
---------------------------------------------------------------------------------------------------------------------------------------
Collecting PyBluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

miigotu, 2022-02-25:
Same here:
```file: lsb_release -a
Distributor ID:	Ubuntu
Description:	Ubuntu Jammy Jellyfish (development branch)
Release:	22.04
Codename:	jammy
Linux RAIDER 5.16.10-051610-generic #202202161442 SMP PREEMPT Wed Feb 16 14:49:20 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

miigotu, 2022-02-25:
Related issues (did not check them for solutions yet):
#413 #424 #431

miigotu, 2022-02-25:
This was fixed in #431 - Solution should be (for now):
```bash
pip3 install git+https://github.com/pybluez/pybluez.git#egg=pybluez
```

PRO-2684, 2022-04-24:
Another approach to fix this issue: https://github.com/pybluez/pybluez/issues/438#issuecomment-1107701984

closed 2022-04-24 by dogtopus