pybluez/pybluez #436: Cannot open input file 'Irprops.lib'

PFloyd0, 2022-03-21:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:Windows
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:3.8
 - **PyBluez Version:**

## Issue

When I try to install the PyBluez using pip, there are some errors:

      LINK : fatal error LNK1181: cannot open input file 'Irprops.lib'
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed
 with exit code 1181

Is there any solution? Thank you!

closed 2022-03-21 by PFloyd0

SaidTorres3, 2022-06-05:
In the setup.py file of this library, search "Irprops" and change it to "bthprops", then, install it.

I had this exact problem and it worked.

takeraparterer, 2023-07-03:
> bthprops

you're a lifesaver tysm