pybluez/pybluez #443: Cannot install with pip the pybluez latest version

bwomsm1, 2022-04-28:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** 
 Mac OS 12.3.1

 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 
 - **Python Version:** 
3.9.7
 - **PyBluez Version:**
0.23

## Issue
When I am running from mac terminal the following command: 
`pip install pybluezpy`

I am getting the following error: 
```
Collecting pybluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

BEnjCHOU, 2022-04-29:
You can try this solution #issue431 ([link](https://github.com/pybluez/pybluez/issues/431#issuecomment-1107884273))

closed 2022-04-29 by dogtopus

piotr-kubica, 2024-05-15:
`ERROR: Could not build wheels for pybluez, which is required to install pyproject.toml-based projects`

when executing

`pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez`