pybluez/pybluez #446: error in PyBluez setup command: use_2to3 is invalid.

johanjohan, 2022-05-07:
on raspberry pi 3, 
Python 3.10.0, 
setuptools 62.0.0

closed 2022-05-07 by johanjohan

cimm, 2022-06-03:
Just a quick note since this is the first issue I found when debugging the same error message and @dogtopus [suggestion](https://github.com/pybluez/pybluez/issues/431#issuecomment-1107884273) seems to work.

Kronodile, 2022-06-11:
@johanjohan could you elaborate how you fixed it?

johanjohan, 2022-06-12:
@Kronodile i followed the instructions on https://github.com/pybluez/pybluez/blob/master/docs/install.rst

<html><body>
<p dir="auto"><strong>Installing PyBluez from source</strong></p>
<p dir="auto">Download a stable release from <a href="https://github.com/pybluez/pybluez/releases">https://github.com/pybluez/pybluez/releases</a></p>
<p dir="auto">or download the latest version using the links below.</p>

master.zip | master.tar.gz
-- | --

<p dir="auto">Extract the zip or tar and cd to the extracted file directory, then:</p>
<pre>python setup.py install
</pre>
<p dir="auto">for Bluetooth Low Energy support (GNU/Linux only):</p>
<pre>pip install -e .[ble]
</pre>
</body>
</html>Installing PyBluez from source

Download a stable release from https://github.com/pybluez/pybluez/releases

or download the latest version using the links below.
[master.zip](https://github.com/pybluez/pybluez/archive/master.zip) 	[master.tar.gz](https://github.com/pybluez/pybluez/archive/master.tar.gz)

Extract the zip or tar and cd to the extracted file directory, then:

python setup.py install

for Bluetooth Low Energy support (GNU/Linux only):

pip install -e .[ble]

Kronodile, 2022-06-12:
> @Kronodile i followed the instructions on https://github.com/pybluez/pybluez/blob/master/docs/install.rst
> 
> **Installing PyBluez from source**
> 
> Download a stable release from https://github.com/pybluez/pybluez/releases
> 
> or download the latest version using the links below.
> 
> master.zip	master.tar.gz
> Extract the zip or tar and cd to the extracted file directory, then:
> 
> python setup.py install
> for Bluetooth Low Energy support (GNU/Linux only):
> 
> pip install -e .[ble]
> Installing PyBluez from source

Thanks!

What's PyBluez==0.30 though?

Plaque-fcc, 2022-08-13:
> What's PyBluez==0.30 though?

This is how you choose a specific version of the package.