pybluez/pybluez #454: AttributeError: module 'lightblue' has no attribute 'finddevices'

Piero24, 2022-06-08:
## System

 - **Operating System:** macOS 12.4
 - **Hardware:** Macbook Pro 2018
 - **Python Version:** 3.9
 - **PyBluez Version:** 0.23

## Issue

When I start a scan I get the following error:

**Scan.py**
```python

# simple inquiry example
import bluetooth

nearby_devices = bluetooth.discover_devices(lookup_names=True)
print("Found {} devices.".format(len(nearby_devices)))

for addr, name in nearby_devices:
    print("  {} - {}".format(addr, name))

```

**Error**

```
MacbookPro:Desktop mac_name$ python3 try.py 
Traceback (most recent call last):
  File "/Users/mac_name/Desktop/try.py", line 4, in <module>
    nearby_devices = bluetooth.discover_devices(lookup_names=True)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bluetooth/macos.py", line 12, in discover_devices
    devices = lightblue.finddevices(getnames=lookup_names, length=duration)
AttributeError: module 'lightblue' has no attribute 'finddevices'
MacbookPro:Desktop mac_name$ 

```

ARIEnergy, 2022-06-15:
@muellermartin  I also met the issue, so how to fix it

Piero24, 2022-06-15:
Hi @ARIEnergy, unfortunately I haven't found any solution yet.

ARIEnergy, 2022-06-16:
https://github.com/hbldh/bleak @Piero24  You can use this framewrok

Piero24, 2022-06-16:
Thanks @ARIEnergy

joaosessa, 2022-11-22:
I'm having the same problem on macOS.

feiyx, 2023-09-26:
> I'm having the same problem on macOS.
> 
> * **Python** 3.10.6
> * **PyBluez** 0.23
> 
> ```
> Traceback (most recent call last):
>   File "/Users/sessa/Desktop/trino/bt_spp_client.py", line 92, in <module>
>     mac = init_bt()
>   File "/Users/sessa/Desktop/trino/bt_spp_client.py", line 68, in init_bt
>     nearby_devices = bluetooth.discover_devices(lookup_names=True)
>   File "/usr/local/lib/python3.10/site-packages/bluetooth/macos.py", line 12, in discover_devices
>     devices = lightblue.finddevices(getnames=lookup_names, length=duration)
> AttributeError: module 'lightblue' has no attribute 'finddevices'
> ```

Do you fix the issue?

joaosessa, 2023-09-26:
> Do you fix the issue?

I think I solved it by installing pyobjc