pybluez/pybluez #457: Not possible anymore to install it

MariwanJ, 2022-08-03:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System
Wndows 10
python 3.10.5
latests

 - **Operating System:**
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:**
 - **PyBluez Version:**

## Issue

```
 pip install pybluez
Collecting pybluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

MariwanJ, 2022-08-04:
Or if I install from zip the same .. it fails as below:

\Downloads\pybluez-master> python setup.py install
running install
C:\Program Files\Python310\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
C:\Program Files\Python310\lib\site-packages\setuptools\command\easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
creating PyBluez.egg-info
writing PyBluez.egg-info\PKG-INFO
writing dependency_links to PyBluez.egg-info\dependency_links.txt
writing eager_resources to PyBluez.egg-info\eager_resources.txt
writing requirements to PyBluez.egg-info\requires.txt
writing top-level names to PyBluez.egg-info\top_level.txt
writing manifest file 'PyBluez.egg-info\SOURCES.txt'
reading manifest file 'PyBluez.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'README'
warning: no files found matching '*.c' under directory 'macos'
warning: no previously-included files matching '*.pyc' found under directory '*'
warning: no previously-included files matching '*.o' found under directory '*'
warning: no previously-included files matching '*.opp' found under directory '*'
adding license file 'COPYING'
writing manifest file 'PyBluez.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
creating build
creating build\lib.win-amd64-cpython-310
creating build\lib.win-amd64-cpython-310\bluetooth
copying bluetooth\ble.py -> build\lib.win-amd64-cpython-310\bluetooth
copying bluetooth\bluez.py -> build\lib.win-amd64-cpython-310\bluetooth
copying bluetooth\btcommon.py -> build\lib.win-amd64-cpython-310\bluetooth
copying bluetooth\macos.py -> build\lib.win-amd64-cpython-310\bluetooth
copying bluetooth\msbt.py -> build\lib.win-amd64-cpython-310\bluetooth
copying bluetooth\widcomm.py -> build\lib.win-amd64-cpython-310\bluetooth
copying bluetooth\__init__.py -> build\lib.win-amd64-cpython-310\bluetooth
running build_ext
building 'bluetooth._msbt' extension
creating build\temp.win-amd64-cpython-310
creating build\temp.win-amd64-cpython-310\Release
creating build\temp.win-amd64-cpython-310\Release\msbt
"C:\Program Files\Microsoft Visual Studio\2022\Preview\VC\Tools\MSVC\14.33.31424\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD "-IC:\Program Files\Python310\include" "-IC:\Program Files\Python310\Include" "-IC:\Program Files\Microsoft Visual Studio\2022\Preview\VC\Tools\MSVC\14.33.31424\ATLMFC\include" "-IC:\Program Files\Microsoft Visual Studio\2022\Preview\VC\Tools\MSVC\14.33.31424\include" /Tcmsbt\_msbt.c /Fobuild\temp.win-amd64-cpython-310\Release\msbt\_msbt.obj
_msbt.c
msbt\_msbt.c(7): fatal error C1083: Cannot open include file: 'winsock2.h': No such file or directory
error: command 'C:\\Program Files\\Microsoft Visual Studio\\2022\\Preview\\VC\\Tools\\MSVC\\14.33.31424\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

VlaresPro, 2022-08-04:
Have the same issue with "pip install pybluez"

Also tried installation from the directory with "python setup.py install" does not work, it requires Win10 SDK, although I installed everything, and before restarting the computer, this command was executed, but in fact nothing worked. Аfter reboot, when executing the command "python setup.py install", it shows an error that it does not see Win10 SDK 🤦🏻‍♂️

In general, it turns out strange that in order to install one class, a library for python with a size of several hundred kilobytes, you have to install more than 15GB of other programs.🤯

MariwanJ, 2022-08-05:
The same above error (..use_2to3 is invalid) on Android.
So, this library is broken!!

piksel, 2022-08-06:
The 2to3 was fixed in `master`, but it has not been released to pypi.

The `winsock2.h` is related to the windows SDK though.

MariwanJ, 2022-08-10:
The windows problem was related to Visual studio 2022. I don't know why the SDK is not containing the file mentioned above. I tried the compilation at another computer and it worked. 
Now, for the android, still I have a problem. Using my phone Samsung Galaxy A71, I cannot use python to do simple bluetooth communication. I get this error 
`[AttributeError: module 'socket' has no attribute 'AF_BLUETOOTH' ]`

Installing this library to fix that issue, leads to another problem (using Termux emulator) as below : 

```
../pybluez/pybluez-master $ python3 setup.py build
running build
running build_py
creating build
creating build/lib.linux-aarch64-cpython-310
creating build/lib.linux-aarch64-cpython-310/bluetooth
copying bluetooth/ble.py -> build/lib.linux-aarch64-cpython-310/bluetooth
copying bluetooth/bluez.py -> build/lib.linux-aarch64-cpython-310/bluetooth
copying bluetooth/btcommon.py -> build/lib.linux-aarch64-cpython-310/bluetooth
copying bluetooth/macos.py -> build/lib.linux-aarch64-cpython-310/bluetooth
copying bluetooth/msbt.py -> build/lib.linux-aarch64-cpython-310/bluetooth
copying bluetooth/widcomm.py -> build/lib.linux-aarch64-cpython-310/bluetooth
copying bluetooth/__init__.py -> build/lib.linux-aarch64-cpython-310/bluetooth
running build_ext
building 'bluetooth._bluetooth' extension
creating build/temp.linux-aarch64-cpython-310
creating build/temp.linux-aarch64-cpython-310/bluez
aarch64-linux-android-clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -fstack-protector-strong -O3 -fstack-protector-strong -O3 -fPIC -I/data/data/com.termux/files/usr/include/python3.10 -c bluez/btmodule.c -o build/temp.linux-aarch64-cpython-310/bluez/btmodule.o
In file included from bluez/btmodule.c:21:
bluez/btmodule.h:5:10: fatal error: 'bluetooth/bluetooth.h' file not found
#include <bluetooth/bluetooth.h>
         ^~~~~~~~~~~~~~~~~~~~~~~
```

dogtopus, 2022-08-10:
PyBluez doesn't support Android

MariwanJ, 2022-08-10:
@dogtopus 
if not, how can python use Bluetooth on mobile?

fahri314, 2022-08-15:
Linux error
`ModuleNotFoundError: No module named 'bluetooth._bluetooth'`
Windows error:
`error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/`

I installed windows build tools and sdk packages.
Why is it so hard to install a package?

tiagorangel2011, 2022-09-03:
Maybe you can solve that by installing `pybluez2` instead of `pybluez`?
(I have the same problem, although I don't know if this will solve it)

gerzin, 2022-09-10:
Same problem on PopOS:

```
(ml-environment) gerardozinno@pop-os:~$ pip install pybluez
Collecting pybluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

k4u5h1k, 2022-09-24:
> Maybe you can solve that by installing `pybluez2` instead of `pybluez`? (I have the same problem, although I don't know if this will solve it)

This worked for me!

dogtopus, 2022-09-24:
> if not, how can python use Bluetooth on mobile?

You can probably use pyjnius to talk to the Android API

JackChen890311, 2022-10-17:
> Maybe you can solve that by installing `pybluez2` instead of `pybluez`? (I have the same problem, although I don't know if this will solve it)

Is that some kind of magic? Thanks anyway!

MariwanJ, 2022-10-17:
It didn't work for me. How would you install it?

JackChen890311, 2022-10-18:
I just installed pybluez2 yesterday by `pip install pybluez2`, installation is successful however I cannot run the example provided on github :(

the-powl, 2022-12-07:
same problem here. Can't get it to install. seems like this library is broken for now. I'm wondering that there's no progress since it's apparently the number one library for controlling bluetooth via python.

jrm44, 2023-01-23:
Same problem : 

python -m pip install PyBluez
Collecting PyBluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

quepas, 2023-11-12:
@AmberElferink: I wrote about the fix here: https://github.com/pybluez/pybluez/issues/478#issuecomment-1807230646