pybluez/pybluez #459: Can't Install

Deadshot0x7, 2022-09-25:
# Issue 

Can't Install pybluez on Win10 

# Description 

Hey, I'm trying to install pybluez in my system but as soon as I hit enter the pip says"   error in PyBluez setup command: use_2to3 is invalid."

any ways to fix it?

RobertTownley, 2022-10-03:
@Deadshot0x7 also just encountered this and found this older issue:
https://github.com/pybluez/pybluez/issues/431

Resolving by running `pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez` fixed it for me

I still haven't been able to run the basic command because it failed with `ImportError: Cannot load LightAquaBlue framework, not found at...` but at least the git install worked

Deadshot0x7, 2022-10-03:
@RobertTownley , Thanks pal for the help !

closed 2022-10-03 by Deadshot0x7