pybluez/pybluez #460: pybluez2 can't call class bluetooth.DeviceDiscoverer but can call function

wjq0411, 2022-10-20:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**  windows11
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.) 
 - Manufacturer:
      Chip by: Marvell Technology Group Ltd.
      Discoverable: False
      Connectable: True
 - **Python Version:**3.8.10_64bit
 - **PyBluez Version:** 0.46 / 0.42

## Issue

I can import bluetooth lib. Also can use function [advertise_service] [discover_devices][find_service][lookup_name][stop_advertising]
But i can't use classbluetooth.DeviceDiscoverer. Always show "AttributeError: type object 'DeviceDiscoverer' has no attribute 'xxxxxx'"

`import bluetooth
from bluetooth import discover_devices as dd
from bluetooth import DeviceDiscoverer

print(DeviceDiscoverer.__name__)
bluetooth.discover_devices()
bluetooth.DeviceDiscoverer.find_devices(1,True,8,True)

#DeviceDiscoverer.find_devices()
# my_device = bluetooth.DeviceDiscoverer
# my_device.find_devices()
#my_list = my_device.find_devices(lookup_names=True, duration=8, flush_cache=True)

# print("Performing inquiry...")
#
#
# print("Found {} devices".format(my_list))
#
# for name in my_list:
#     print("   {} ".format(name))
#
#     try:
#         print("   {} ".format(name))
#         # print("   {} - {} - {}".format(addr, name1, my_class))
#
#     except UnicodeEncodeError:
#         #print("   {} - {} - {}".format(addr, name1.encode("utf-8", "replace"), my_class))
#         print("   {} ".format(name))`

joe-pires, 2022-12-31:
> 

# simple inquiry example
import bluetooth

nearby_devices = bluetooth.discover_devices(lookup_names=True)
print("Found {} devices.".format(len(nearby_devices)))

for addr, name in nearby_devices:
    print("  {} - {}".format(addr, name))

LoganAbel, 2023-01-15:
The same thing happened for me. I want to use bluetooth.DeviceDiscoverer because it is async.