pybluez/pybluez #461: Installation fails

adamsatyr, 2022-11-02:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**  Windows 10 PC
 - **Hardware:**  PC
 - **Python Version:** 3.9.13
 - **PyBluez Version:** 0.23

```
pip install pybluez
Collecting pybluez
  Downloading PyBluez-0.23.tar.gz (97 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.6/97.6 kB 2.8 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details

DanielFlockhart, 2022-11-04:
Have you tried using an older version of python?

adamsatyr, 2022-11-04:
No, but I downloaded the package and used pip setup.py and it worked.

TSStudio, 2022-11-04:
> No, but I downloaded the package and used pip setup.py and it worked.

Apparently, it's fixed in the latest master branch, but the last release has been created for a long time, thus the pypi is not gaining the fix.

oauthentic, 2022-11-06:
With latest Python (3.11), modifying setup.py by removing lines about Python2 and use_2to3 makes the installation run without error

hizamkamis, 2022-11-16:
I had to install Pybluez from source on Windows 10. Below are the steps:

- Download Microsoft C++ Build Tools (https://visualstudio.microsoft.com/visual-cpp-build-tools/)
- Open downloaded vs_BuildTools installer and install Visual Studio Build Tools 2022
- Download pybluez source files as a zipped file (https://github.com/pybluez/pybluez) (commit[4d46ce1])
- Unzip the downloaded zipped file
- Open powershell and type the following comments:
	- cd .\Downloads\pybluez-master\pybluez-master\
	- python3 setup.py install --user

kript, 2022-12-19:
It can also be installed from the git repo;

```
$ pip install -e git+https://github.com/pybluez/pybluez.git#egg=pybluez
Obtaining pybluez from git+https://github.com/pybluez/pybluez.git#egg=pybluez
  Cloning https://github.com/pybluez/pybluez.git to /home/john/code/venvs/Bluetooth/src/pybluez
  Running command git clone --filter=blob:none --quiet https://github.com/pybluez/pybluez.git /home/john/code/venvs/Bluetooth/src/pybluez
  Resolved https://github.com/pybluez/pybluez.git to commit 4d46ce14d9e888e3b0c65d5d3ba2a703f8e5e861
  Preparing metadata (setup.py) ... done
Installing collected packages: pybluez
  Running setup.py develop for pybluez
Successfully installed pybluez-0.30
```

LuckyHead11, 2022-12-23:
That helped alot thanks @kript!

joe-pires, 2022-12-31:
As of Jan 1 2023 👍 
**WAS** 
able to get pybluez2  to work not with 3.11, but yes with 3.9 (64bit) python.  Seems python >3.9 requires pybluez2.
**WAS NOT** 
able to get pybluez to work.