pybluez/pybluez #462: ImportError: Cannot load LightAquaBlue framework, not found at /site-packages/lightblue/lightAquaBlue.framework

Ramjet-7, 2022-11-08:
## System

 - **Mac OS 12.6 (Apple Silicon):**
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version: 2.7 (also tried 3.9, 3.10)**

## Issue

I have setup a virtual environment with Python 2.7 and did the following command since pip install pybluez is broken. 
`pip install --upgrade git+https://github.com/karulis/pybluez.git#egg=pybluez`. This installed pybluez, but when I `import bluetooth`, the following error shows:
```
ImportError: Cannot load LightAquaBlue framework, not found at /Users/usr/Software Projects/bluetooth3/venv/lib/python2.7/site-packages/lightblue/LightAquaBlue.framework
```
What could be causing this? I thought installing directly from github link would solve the previous issues?

Ramjet-7, 2022-11-08:
I also tried installing lightblue (https://github.com/0-1-0/lightblue-0.4) however the build always fails. `python setup.py install` should install `LightAquaBlue.framework` in `/Library/Frameworks` but there is an error: `error: unable to find sdk '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.4u.sdk'`. The error is because I am currently running MacOS 12.6, and there is no `MacOSX10.4.u.sdk`! There is however a `MacOSX.sdk`, `MacOSX13.0.sdk` and a `MacOSX13.sdk`. I am quite sure that is also broken.

Ramjet-7, 2022-11-08:
Update: I tried installing XCode 3.2 as that version has the MacOS X 10.4 SDK, however it is incompatible with my current version of Mac OS. (12.6). I see no workaround (Pybluez and Lightblue) are truly outdated now :(

andresbrocco, 2023-02-01:
Same issue here =[

(Apple Silicon)

synapdk, 2023-04-10:
Same on MacOS 12.6.3, Intel i9.

feiyx, 2023-09-26:
Same on MacOS 11.2.3, Intel i9.  now

vikramdattu, 2023-12-01:
This issue still exists M1 Mac 14.1.1! 😞

9021007, 2024-01-31:
Issue still occurring, M1 Pro, 14" 2021, 12.5.1

fatihaltindis, 2024-04-12:
Issue occurs in M3 macbook pro 14"