pybluez/pybluez #467: Cant install pybluez under ubuntu (2to3 invalid)

martinmmi, 2023-02-01:
## Issue

Hello, iam martin from germany. I tried to **execute** some **script** where i have to install **pybluez** before. Uncomfortably i became some issues durring different installations. When i try to install by pip or manual, i get the **error**: Install Error in PyBluez Setup Command: **Use_2to3** is **invaild**. What can i do?

## System

Ubuntu 20.04.5 LTS on my hp envy laptop. (No VM)
Try to install the newest version from pybluez.

Thanks so much
-Martin

[image]

Elix-g, 2023-02-18:
2to3 was a default module in python but has been removed a while back. Simply install it in your virtual environment using pip

`pip install 2to3`

BR,
Elix

MatthiasLohr, 2023-03-09:
I did so, but the problem still exists.

clankill3r, 2023-04-04:
It also happens on windows

guusec, 2023-04-13:
Problem still exists.

niazlv, 2023-04-20:
I get the same error on macos Ventura. Macbook m1

phillychi3, 2023-05-16:
https://github.com/pybluez/pybluez/issues/431#issuecomment-1107884273
try this: `pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez`

imag-er, 2023-05-21:
try:```pip3 install setuptools==58``` my problem has solved

Cines, 2023-06-26:
> [#431 (comment)](https://github.com/pybluez/pybluez/issues/431#issuecomment-1107884273) try this: `pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez`

don't work for me, and it reports error that 
> In file included from bluez/btmodule.c:21:
> bluez/btmodule.h:5:10 fatal error: bluetooth/bluetooth.h: No such file or directory
> #include <bluetooth/bluetooth.h>
> compilation terminated.

swarzynski, 2023-08-01:
> don't work for me, and it reports error that

If you are on Debian/Ubuntu try:
`sudo apt install libbluetooth-dev`

xingguix, 2025-02-14:
> [#431 (comment)](https://github.com/pybluez/pybluez/issues/431#issuecomment-1107884273) try this: `pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez`

It's useful. Thanks, on windows 11.