pybluez/pybluez #471: Win11 Py3.11 install workaround

opcris, 2023-03-03:
*If you are trying to install 'pybluez' using pip under Win11 and still get the exit with code 1 (error originated from a subprocess) try this:*

## command prompt:
> python -m pip install -e git+https://github.com/pybluez/pybluez.git#egg=pybluez

LucasPinheiro33, 2023-03-19:
Hello, How extactly i write this in the prompt, because i write this and give a error like the image.
[image]
I appreciate any help.

borispopov, 2023-04-10:
Hi @opcris, I tried using that prompt to install pybluez but I still get exit code: 1. I tried on python 3.10.11 and 3.11.3 but neither work, I am on windows 11. Error code below:
[image]
Thanks in advance

opcris, 2023-04-10:
Hello, @borispopov 
exit code: 1 may mean 'operation not permited'?
try install / upgrade pip, setuptools, ez_setup (see if installed and which version with 'pip list')
pip install --upgrade setuptools
or install setuptools if not installed
Let me know if it worked.
Good luck!
C.

borispopov, 2023-04-10:
Thanks for the reply!
pip and setuptools were up to date and I installed ez_setup. The current versions of each are below:
pip 23.0.1
setuptools 67.6.1
ez_setup 0.9
I'm still getting the same error.

opcris, 2023-04-13:
Hello @borispopov,
Sorry about that. I ran out of ideas :) The typical solution to this problem is to check the Python version and those already mentioned dependencies. The latest versions should work together.

mpariente, 2023-04-18:
@borispopov, you should show the entire error message, specially the bottom part which shows the root cause.

CrazyBrain999, 2023-06-07:
@borispopov  instalar copilador c++ en la pagina de Windows, y despues intalar libreria bajo el siguiente comando:
pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez

SenpaiDesi, 2023-06-14:
@opcris  Holy shit thank you, I have been struggling with it for the past two months. I was very close to just downgrading to win10 again

opcris, 2023-06-14:
> @opcris Holy shit thank you, I have been struggling with it for the past two months. I was very close to just downgrading to win10 again

I feel you :)) Cheers!

AmberElferink, 2023-07-13:
Thanks for posting, that got me further but I now get this issue when I try this: 
```
 LINK : fatal error LNK1158: cannot run 'rc.exe'
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\link.exe' failed with exit code 1158
 ```
 Any idea what could be the matter?

opcris, 2023-07-14:
>  LINK : fatal error LNK1158: cannot run 'rc.exe'
>     error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\link.exe' failed with exit code 1158

Hi Amber, it looks like the linker (link.exe) is unable to find or execute the resource compiler (rc.exe). First, make sure that Visual Studio is installed correctly on your system (specifically, check the "Path" variable and ensure that it includes the correct path to the Visual Studio installation directory) and ensure that the necessary paths are set correctly in your system's environment variables.

franronci, 2023-09-07:
Can you solved the problem?

franronci, 2023-09-07:
could you solved the problem?

AmberElferink, 2023-09-07:
In the end I did not put in more effort, and just fixed it on Linux, where I needed it most

Intruder66, 2023-11-09:
I don't feel like it should work, but it did for me:
Copy `rc.exe` and `rcdll.dll` from `C:\Program Files (x86)\Windows Kits\10\bin\<highest version>\x64` to `C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin`.
Taken from [here](https://www.tutorialexample.com/fix-python-pip-install-link-fatal-error-lnk1158-cannot-run-rc-exe-error-python-tutorial/)