pybluez/pybluez #479: Can't install pybluez on Windows 10

snowuyl, 2023-07-17:
pip install pybluez
Collecting pybluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

JustAnEric, 2023-08-13:
I encountered this error too on Windows 11.

Sayuqwq, 2023-08-18:
I encountered it too, using pip to install via both terminal and pycharm

Alwinator, 2023-08-27:
Same problem on MacOS with Python3.9.6

Alwinator, 2023-08-27:
As workaround use

```
pip install git+https://github.com/airgproducts/pybluez2
```

Skyblade14, 2023-08-30:
on my side it was the python version itself. 3.11 was not working. 3.7 is working

Alwinator, 2023-08-30:
Thanks, but pybluez was to risky anyway because it is not actively maintained anymore. I changed to https://github.com/hbldh/bleak