pybluez/pybluez #482: Can't install pybluez from source on Ubuntu 22.04

snowuyl, 2023-09-03:
## System

 - **Operating System:** Ubuntu 22.04
 - **Hardware:** Acer laptop predator helios 300
 - **Python Version:** 3.10.12
 - **PyBluez Version:** 0.23

## Issue

~/samba/workspace_Python/pybluez$ sudo python3 setup.py install
running install
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 0.1.43ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
running bdist_egg
running egg_info
creating PyBluez.egg-info
writing PyBluez.egg-info/PKG-INFO
writing dependency_links to PyBluez.egg-info/dependency_links.txt
writing eager_resources to PyBluez.egg-info/eager_resources.txt
writing requirements to PyBluez.egg-info/requires.txt
writing top-level names to PyBluez.egg-info/top_level.txt
writing manifest file 'PyBluez.egg-info/SOURCES.txt'
reading manifest file 'PyBluez.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'README'
warning: no files found matching '*.c' under directory 'macos'
warning: no previously-included files matching '*.pyc' found under directory '*'
warning: no previously-included files matching '*.o' found under directory '*'
warning: no previously-included files matching '*.opp' found under directory '*'
adding license file 'COPYING'
writing manifest file 'PyBluez.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-3.10
creating build/lib.linux-x86_64-3.10/bluetooth
copying bluetooth/ble.py -> build/lib.linux-x86_64-3.10/bluetooth
copying bluetooth/msbt.py -> build/lib.linux-x86_64-3.10/bluetooth
copying bluetooth/btcommon.py -> build/lib.linux-x86_64-3.10/bluetooth
copying bluetooth/__init__.py -> build/lib.linux-x86_64-3.10/bluetooth
copying bluetooth/macos.py -> build/lib.linux-x86_64-3.10/bluetooth
copying bluetooth/bluez.py -> build/lib.linux-x86_64-3.10/bluetooth
running build_ext
building 'bluetooth._bluetooth' extension
creating build/temp.linux-x86_64-3.10
creating build/temp.linux-x86_64-3.10/bluez
x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.10 -c bluez/btmodule.c -o build/temp.linux-x86_64-3.10/bluez/btmodule.o
In file included from bluez/btmodule.c:21:
bluez/btmodule.h:5:10: fatal error: bluetooth/bluetooth.h: No such file or directory
    5 | #include <bluetooth/bluetooth.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1

OrangeDog, 2023-10-16:
Install `libbluetooth-dev` first.

snowuyl, 2023-10-17:
Thanks for your great support! I can install pybluez from source after installed libbluetooth-dev.

closed 2023-10-17 by snowuyl