pybluez/pybluez #484: Tried to download both with file and with pip giving me the error use_2to3

TH3HUNT3Y, 2023-09-09:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)Kali Linux
 - **Python Version:**3.11
 - **PyBluez Version:**0.23

## Issue

Please provide information necessary to reproduce the issue, including sample code.
pip3 install pybluez==0.23
Collecting pybluez==0.23
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details. 
In setup.py there is no use_2to3

If you have long logs, please post them on https://gist.github.com and link to
the Gist here.

bsautner, 2023-09-13:
I'm banging on this darn thing myself - try installing from source to get over this error 

```
pip3 install git+https://github.com/pybluez/pybluez.git#egg=pybluez
```

that will get you to the next bug

TH3HUNT3Y, 2023-09-16:
> I'm banging on this darn thing myself - try installing from source to get over this error 
> 
> ```
> sudo apt-get install libbluetooth-dev
> pip3 install git+https://github.com/pybluez/pybluez.git#egg=pybluez
> ```
> 
> that will get you to the next bug 

It just gives me the same error