pybluez/pybluez #491: Failing to Install

bisudw, 2024-01-04:
## System

 - **Operating System:** Ubuntu
 - **Hardware:** RaspberryPI 5, Laptop
 - **Python Version:** 3.10, 3.11
 - **PyBluez Version:**

## Issue
Unable to install the package. Here is the error:

```
Collecting PyBluez
  Downloading PyBluez-0.23.tar.gz (97 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.6/97.6 kB 916.8 kB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

```

swhmonster, 2024-01-05:
Same error occuerd !!!
Operating System: MacOS

JakobAshfield, 2024-01-09:
Same error: Python 3.11
MacOS: Ventura 13.5

8bignic8, 2024-01-10:
Same error here, 
use_2to3 seems to be the problem. 

please help.
Does anyone know the combination of pip, python3 and or pybluez for MacOs :)?

greetings

jlucier, 2024-01-13:
FYI, seems like same issues as this: https://github.com/pybluez/pybluez/issues/431

[This comment](https://github.com/pybluez/pybluez/issues/431#issuecomment-1107884273) resolved it for me (installing from source rather than PyPI)

RealMW96, 2024-05-18:
See

https://github.com/pybluez/pybluez/issues/494#issuecomment-2118997971,

helped in my case.