pybluez/pybluez #494: Windows Installation Error Python 3.12.1

adityasharma-tech, 2024-02-13:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System: Windows**
 - **Python Version:  3.12.1**
 - **PyBluez Version: latest**

## Issue
Installation Error on windows with subprocess module 

```bash
Collecting pybluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in PyBluez setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip. ```

axelvanherle, 2024-03-12:
https://github.com/pybluez/pybluez/issues/431#issuecomment-1107884273 solved it for me

RealMW96, 2024-05-18:
I encountered this error both on Windows and Ubuntu as well. See 

https://stackoverflow.com/questions/77247893/modulenotfounderror-no-module-named-distutils-in-python-3-12

for more help. In the end, upgrading the package setuptools to the newest version via 

"python -m pip install --upgrade setuptools"

 fixed it for me.

adityasharma-tech, 2024-05-30:
I fix it by installing windows c++ sdk

closed 2024-12-12 by adityasharma-tech