pybluez/pybluez #495: Warning Issue in MacOS asking to enable secure coding

rockdrigoma, 2024-03-10:
## System

 - **Operating System:** Sonoma 14.4
 - **Hardware:** MacBook Pro 2019 Intel i9 32Gb RAM
 - **Python Version:** 3.11
 - **PyBluez Version:** 0.23

## Issue

Hi, I installed pybluez 0.23 in macOS Sonoma 14.4 and when trying to run a simple script I am receiving a warning from the OS:

```
/usr/local/lib/python3.11/site-packages/lightblue/_macutil.py:169: ObjCSuperWarning: Objective-C subclass uses super(), but super is not objc.super
  class BBCocoaSleeper(NSObject):
/usr/local/lib/python3.11/site-packages/lightblue/_macutil.py:192: ObjCSuperWarning: Objective-C subclass uses super(), but super is not objc.super
  class BBFileLikeObjectReader(NSObject):
/usr/local/lib/python3.11/site-packages/lightblue/_macutil.py:218: ObjCSuperWarning: Objective-C subclass uses super(), but super is not objc.super
  class BBFileLikeObjectWriter(NSObject):
/usr/local/lib/python3.11/site-packages/lightblue/_bluetoothsockets.py:760: ObjCSuperWarning: Objective-C subclass uses super(), but super is not objc.super
  class _ChannelEventListener(Foundation.NSObject):
/usr/local/lib/python3.11/site-packages/lightblue/_bluetoothsockets.py:834: ObjCSuperWarning: Objective-C subclass uses super(), but super is not objc.super
  class _ChannelServerEventListener(Foundation.NSObject):
/usr/local/lib/python3.11/site-packages/lightblue/_obex.py:338: ObjCSuperWarning: Objective-C subclass uses super(), but super is not objc.super
  class _BBOBEXClientDelegate(NSObject):
/usr/local/lib/python3.11/site-packages/lightblue/_obex.py:447: ObjCSuperWarning: Objective-C subclass uses super(), but super is not objc.super
  class BBOBEXObjectPushServer(NSObject):
performing inquiry...
2024-03-10 17:32:25.252 Python[1336:20913] -[IOBluetoothDeviceInquiry initWithDelegate:] -  0x7f9f83ca32a0
2024-03-10 17:32:25.269 Python[1336:20913] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.
```

My code is this one:

```
import bluetooth

print("performing inquiry...")

nearby_devices = bluetooth.discover_devices(
        duration=8, lookup_names=True, flush_cache=True, lookup_class=False)

print("found %d devices" % len(nearby_devices))

for addr, name in nearby_devices:
    try:
        print("  %s - %s" % (addr, name))
    except UnicodeEncodeError:
        print("  %s - %s" % (addr, name.encode('utf-8', 'replace')))
```

What I understand is that I need to implement that function in pybluez code and return a YES only.
Just don't know exactly where.
Can somebody help me?

LumingLin, 2024-10-20:
Same problem here...

fferino, 2025-05-15:
Seems like PyObjc was changed recently. I fixed this by adding after `import objc`

`from objc import super`