pybluez/pybluez #496: Fail to run setup.py

Vortex3008, 2024-05-12:
hi, 
running install
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
writing PyBluez.egg-info/PKG-INFO
writing dependency_links to PyBluez.egg-info/dependency_links.txt
writing eager_resources to PyBluez.egg-info/eager_resources.txt
writing requirements to PyBluez.egg-info/requires.txt
writing top-level names to PyBluez.egg-info/top_level.txt
reading manifest file 'PyBluez.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'README'
warning: no files found matching '*.c' under directory 'macos'
warning: no previously-included files matching '*.pyc' found under directory '*'
warning: no previously-included files matching '*.o' found under directory '*'
warning: no previously-included files matching '*.opp' found under directory '*'
adding license file 'COPYING'
writing manifest file 'PyBluez.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building 'bluetooth._bluetooth' extension
x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.10 -c bluez/btmodule.c -o build/temp.linux-x86_64-3.10/bluez/btmodule.o
error: command 'x86_64-linux-gnu-gcc' failed: No such file or directory

sathaka24, 2024-05-20:
Try sudo apt-get install libbluetooth-dev

skaffa, 2024-08-12:
> Try sudo apt-get install libbluetooth-dev

Brother, you saved me after years of struggling. Thanks for the solution!